Guilouz/Creality-Helper-Script-Wiki #3: Z offset ist not permanent

Mavy1704, 2023-08-19:
The Z-offset is transferred to the printer.cfg after "save config", but is not used when printing.  The first layer is always very thin and wide.  I've already done SSH Connection and Permanent Moonraker Configuration.  After turning on the printer, I have to keep re-entering the Z offset for it to work.  But I would like to save it permanently.  Where is the problem?  Can someone help?

Guilouz, 2023-08-19:
Please don't create issue for non relative Repo issue.

You must save to Endstop your Gcode offset first on web interface and then apply a SAVE_CONFIG.

closed 2023-08-19 by Guilouz

Mavy1704, 2023-08-19:
OK, understood ;-)

That's what I did because Klipper wanted it that way, but it never worked. Hence the question. I've now written it into the START-PRINT macro....SET_GCODE_OFFSET Z=-0.06. That's the only way it works.