Guilouz/Creality-Helper-Script-Wiki #4: Inaccuracies/outdated information in repository, "missing" profiles

gschmidl, 2023-08-19:
1) On the "Update Fluidd" wiki page, you write

> ⚠ DO NOT UPDATE MOONRAKER VERSION AND DO NOT RECOVER KLIPPER VERSION, ONLY FLUIDD!

but it's updating Klipper and recovering Moonraker in the screenshots.

2) In the Readme for the OrcaSlicer profile, there is no full information on which of the Textures files does which. The PNGs all seem identical to me, so it might not matter.

3) Could you add a 0.24mm and 0.28 profile for OrcaSlicer as well? Thank you.

Guilouz, 2023-08-20:
Klipper shouldn't be updated because it's a modified Creality build (although I think it's blocked, I haven't tried) and moonraker shouldn't be updated via the update manager.

I have edited profile files and removed unused texture (I have updated readme file).

I have also added profile for 0.24mm and 0.28mm.

closed 2023-08-20 by Guilouz

gschmidl, 2023-08-22:
Sorry, I lost track of this, but it's also not really important. The warning text which reads

> ⚠ DO NOT UPDATE MOONRAKER VERSION AND DO NOT RECOVER KLIPPER VERSION, ONLY FLUIDD!

should read

> ⚠ DO NOT UPDATE KLIPPER VERSION AND DO NOT RECOVER MOONRAKER VERSION, ONLY FLUIDD!

It's clear what you mean, just the text and screenshot show opposite things.

Guilouz, 2023-08-22:
Ah yes! Thanks it's fixed.