Guilouz/Creality-Helper-Script-Wiki #5: Unable to use OrcaProfiles

thuttu, 2023-08-24:
Starting of prints sliced with OrcaSlicer + profiles fails to errors 

10:21:47 
!! {"code":"key60", "msg":"Internal error on command:CX_PRINT_DRAW_ONE_LINE", "values": ["CX_PRINT_DRAW_ONE_LINE"]}
10:21:47 
!! {"code":"key60", "msg":"Internal error on command:START_PRINT", "values": ["START_PRINT"]}
10:42:20 
!! {"code":"key60", "msg":"Internal error on command:CX_PRINT_DRAW_ONE_LINE", "values": ["CX_PRINT_DRAW_ONE_LINE"]}
10:43:50 
$ RESTART
10:43:50 
// Klipper state: Disconnect

Guilouz, 2023-08-24:
Wich fw version do you use ?

thuttu, 2023-08-24:
I have rooted 1.3.1.19 with all the modifications from wiki, OS is Win 11 and 10 English with Finnish locale

Guilouz, 2023-08-24:
It's look like you have an issue with your START_PRINT macro, check it.

thuttu, 2023-08-24:
Well I as seeing some error somewhere which said that START_PRINT wants %d so decimal values but in gcode they are without decimals (int) also Gcode created with creality print works just fine.

Guilouz, 2023-08-24:
I only use OrcaSlicer and it’s working fine.

closed 2023-08-27 by Guilouz

thuttu, 2023-08-29:
Finally got time to debug with printer it seems that if I upload and start print it fails, if I start print from fluidd it fails.
But if I start print from printers touch screen it works just fine.

Guilouz, 2023-08-29:
Read the wiki…

thuttu, 2023-08-29:
I just realized that there is 3 lines missing from start print.