Guilouz/Creality-Helper-Script-Wiki #6: Timelapse issue

Crazydoub, 2023-08-28:
Hi dear, thanks for your great job.
When installing CFW, original timelapse function doesn't work anymore.
After printing, there's no video file in videos list.
I tried to install mooraker-timelapse but, timelapse icon doesn't appear in fluidd interface.
Is there a way to use CFW and keep original timelapse function (even if we need to transfer video to usb before usin it)
Best regards.

Guilouz, 2023-08-28:
Custom Firmware does not change anything in functionality, they just give root access and that's it. Stock Timelapses continue to work without issue. So your problem is elsewhere.

Crazydoub, 2023-08-28:
Thanks for fast answer, i think you're french, me too ;)
Yes i followd guide, but now way to activate icon on left menu of fluidd.

Crazydoub, 2023-08-29:
Thanks for information.
When i try to install mainsail, no way to unzip, i have a "No space left on device" error
Is there a way to free space?
Best regards.

Guilouz, 2023-08-29:
Wich printer and motherboard version do you have?

closed 2023-09-02 by Guilouz