Guilouz/Creality-Helper-Script-Wiki #7: Update Klipper?

AiramsRT, 2023-09-03:
Hello.  

First of all, thank you for the good work you have done rooting the Creality K1! I installed it yesterday and it worked great.  I have a question, I have seen that klipper is in version 0.9 on the printer, is it not possible to update it?  Especially to be able to use KAMP and Exclude objet which is only possible in Klipper 0.11.  Thank you !!

Guilouz, 2023-09-03:
Actually it's not possible to use official Klipper build.

closed 2023-09-03 by Guilouz

AiramsRT, 2023-09-04:
Ok Thanks!!

Thrashin, 2023-09-16:
> Actually it's not possible to use official Klipper build.

Actually im pretty sure it is now possible https://github.com/mainsail-crew/creality-k1-klipper if you look through the files in the klipper you will find v0.11