Guilouz/Creality-Helper-Script-Wiki #11: Small error in the Install Official Builds description

Nastras, 2023-09-15:
Hello, thanks in advance for your work and this great wiki! I have already used it many times .-)
I have just discovered that you have added a new script to your repo incl. instructions. Exactly what I was looking for!

There I noticed two small things:

- there is one sh. too much
[image]

To get sftp access i must first:
wget -O -  https://openk1.org/static/entware/install-entware-k1.sh | /bin/sh

and then
opkg install openssh-sftp-server; ln -s /opt/libexec/sftp-server /usr/libexec/sftp-server

only after that I can copy your script to root via the sftp connection.

I hope you can understand what I mean.

Best regards and again a big thank you!

Guilouz, 2023-09-15:
sh it's not an error, it's to give privileges on sh script.

MoboXterm working without SFTP, that was it's not needed to install Entware.
Entware is only for people want to install package.

Yes I will add new functions/information, this is really a first approach.

closed 2023-09-15 by Guilouz

Nastras, 2023-09-15:
Ok that was not known to me. I use another SFTP program and there I can only access the folders when I have installed the share with entware. Thanks for the quick feedback.