Guilouz/Creality-Helper-Script-Wiki #13: Multiple instances of Moonraker managing this repo

Nastras, 2023-09-15:
In the Update Manager, after installing moonraker via your script, I noticed this message here:

[image]

Is it supposed to be like this?

Guilouz, 2023-09-15:
Humm no. I have tested it many times in all directions and I don't know how you managed to get that.

Did you delete the creality builds before installing the official ones?

Nastras, 2023-09-15:
I have reset the printer via the display. Then I installed the new root firmware from the usb stick. Afterwards I checked as root if something was already installed (Moonraker or Mainsail somewhere in /usr/data
There I could not find anything from Moonraker or Mainsail.

After that I copied your script to creality/root and executed step 1 and 3.

After the installation I called Mainsail and saw that in the update manager.

Nastras, 2023-09-15:
I just looked in /usr/share to see if there might be a second Moonraker folder there. Could not find one.

Both Moonraker and Mainsail folders are in /usr/data

Guilouz, 2023-09-15:
I can't reproduce this issue

[image]

Try to remove Moonraker and reinstall it.

Nastras, 2023-09-15:
I have now uninstalled and reinstalled moonraker. The error still exists in the boot manager. I downloaded the script again with the changes you made earlier. I've searched everything but can't find a second moonraker instance on the K1 file service.

I noticed something else. After I reinstalled Moonraker via Oprion 1, I get Fluidd displayed with an error in the update manager, even though I don't use Fluidd and haven't installed it. I think that in the script he pulled a moonraker cdie moonraker file with the update values for fluidd, could that be?

[image]

Nastras, 2023-09-15:
I just checked it and when installing moonraker via option 1 it automatically pulled the moonraker_fluidd.conf.
I think there should be a query here which gui you choose?!

I still have something that's giving me a stomach ache. I don't know why this happens. After restarting the printer via power, it always creates a standard printer.cfg for me.
Do you have an idea why?

Nastras, 2023-09-15:
[image]

Guilouz, 2023-09-15:
The script check if fluidd or mainsail folder is present to copy correct moonraker.conf.

If you have fluidd on update manager that mean you have a fluidd folder in /usr/data/.

Have you rebooted and checking update again after uninstall and reinstall ?

Nastras, 2023-09-15:
The script made the wrong decision for me and took the fluidd config. I don't have fluidd installed. As described above, I did a clean install this morning with the creality root image. I then immediately used your script to install mainsail and moonraker. I didn't use the creality method.

Guilouz, 2023-09-15:
I think you must restore firmware settings first and retry.

Nastras, 2023-09-15:
I just restarted the printer via power and now I have printer.cfg copies again.

[image]

Nastras, 2023-09-15:
Then please tell me in which Rhine sequence I should reinstall?

1. Factory reset on the printer
2. Reinstall firmware
3. Root access
4. use your install script

to forget something?

Nastras, 2023-09-15:
I just found this file and a different uuid is entered here as on the moonraker web interface. This uuid appears to be the second instance displayed in the update manager. Can you check whether this file is available to you and whether the uuid is the same as that in the update manager?

[image]
[image]

Guilouz, 2023-09-15:
> Then please tell me in which order I should reinstall?
> 
> 1. Factory reset on the printer
> 2. Reinstall firmware
> 3. Root access
> 4. use your install script
> 
> to forget something?

Yes that' right

Nastras, 2023-09-15:
Good. 🤘

Do you have any other ideas about the printer.cfg that keeps overwriting itself after a power reset?
And what do you say about the uuid file from above?

Guilouz, 2023-09-15:
If you reset factory it's normal printer.cfg file is overwritted by default one.

Humm no idea, i will investigate, my uuid is different.

Nastras, 2023-09-15:
Yes, that's clear if I do a factory reset the printer.cfg is overwritten. Currently the printer.cfg is overwritten with the factory file after every power restart.

It has to be specified somewhere, like with the moonraker file, that it is not created again?

Guilouz, 2023-09-15:
Have you made some changes in your printer.cfg or have you restored it with a previous version ?

Nastras, 2023-09-15:
Yes, I took my config from the old root image from you. It worked wonderfully there.

I have shared my config as you have already seen for a better overview. Attached is my current printer.cfg which is always overwritten.

`# K1
# Printer_size: 220x220x250
# Version: v1.0.18
# CreateDate: 2023/03/21
# Nozzle_mcu: chip: GD32F303CBT6
#             version: CR-K1-MAX-NOZZLE-V1.0.0
# Leveling_mcu: chip: GD32E230F8P6
#             version: CR-K1-MAX-LEVELING-V1.0.0
# mcu: chip: GD32F303RET6
#      version: CR4CU220812S12
####################################################
######## FOR EXPLOITED/ROOTED MACHINES ONLY ########
####################################################

# Creality K1
# This is a different and more clear organization of the config files.
# Everything has been separated by theme, Like bed leveling, macros, etc.
# It should give a easier time looking for any parameter.
#
# Included config.files

[include bedleveling.cfg]
[include endstops.cfg]
[include extra_macro.cfg]
[include fans.cfg]
[include filamentsensor.cfg]
[include led.cfg]
[include gcode_macro.cfg]
[include mcu.cfg]
[include printer_params.cfg]
[include resonancetester.cfg]
[include sensorless.cfg]
[include steppers.cfg]
[include tmcdrivers.cfg]
#[include timelaspe.cfg] # Uncoment if needed

# printer.cfg

[printer]
kinematics: corexy
max_velocity: 1000
max_accel: 20000
max_accel_to_decel: 10000
max_z_velocity: 20
square_corner_velocity: 5.0
max_z_accel: 300

[bl24c16f]
i2c_mcu: rpi
i2c_bus: i2c.2
i2c_speed: 400000

[idle_timeout]
timeout: 99999999

[virtual_sdcard]
path: /usr/data/printer_data/gcodes

[gcode_arcs]
resolution: 1.0

[pause_resume]
# recover_velocity: 500.

[verify_heater extruder]
[verify_heater heater_bed]
check_gain_time: 120
heating_gain: 1.0
hysteresis: 10

[heater_bed]
heater_pin: PB10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
#control: watermark
min_temp: 0
max_temp: 120

[display_status]

[exclude_object]

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [input_shaper]
#*# shaper_type_x = zv
#*# shaper_freq_x = 63.2
#*# shaper_type_y = mzv
#*# shaper_freq_y = 47.2
#*#
#*# [prtouch default]
#*# z_gap_00 = 0.03
#*# z_gap_01 = 0.03
#*# z_gap_11 = -0.03
#*# z_gap_10 = -0.03
#*# version = 307
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  0.382438, 0.381375, 0.373531, 0.381062, 0.360781
#*# 	  0.351938, 0.354594, 0.367875, 0.329281, 0.307875
#*# 	  0.318719, 0.339375, 0.316219, 0.309844, 0.297156
#*# 	  0.312906, 0.305250, 0.312531, 0.342531, 0.329875
#*# 	  0.340241, 0.319719, 0.301094, 0.309552, 0.321188
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.1
#*# min_x = 5.0
#*# max_x = 215.0
#*# min_y = 5.0
#*# max_y = 215.0
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 27.137
#*# pid_ki = 2.513
#*# pid_kd = 73.270
#*#
#*# [heater_bed]
#*# pid_kp = 66.500
#*# pid_ki = 1.208
#*# pid_kd = 915.203
#*# control = pid
`

Guilouz, 2023-09-15:
Your version at line 3: # Version: v1.0.18 is not correct

That it's checked at startup, actually latest config version # Version: v1.0.22 for K1 Max and # Version: v1.0.19 for K1

Nastras, 2023-09-15:
That was the error for the config. I wasn't aware that the version number was important because it is uncommented.

Thank you very much 😘

Now there is only the error with the second Moonraker instance. So far this has had no effect on anything apart from the error in the update manager.

If you can't think of anything else to do with the second instance, I'll do another reset in a few days, I'm no longer able to do it today 😅

Guilouz, 2023-09-15:
Another people have the same issue like you, i will investigate tonight if i find something.

Nastras, 2023-09-15:
That's good to hear that I'm not alone.
You are the best. Thanks!

Nastras, 2023-09-16:
Hey wanted to ask if you have found something yesterday?

Guilouz, 2023-09-16:
I have updated script and now it moonraker provided by Creality and apply git commands to have official builds.

Can you try ?

Nobbyhexer, 2023-09-16:
I have had the same problem with two instances. Now with the current script it works great.
Fantastic work.

Nastras, 2023-09-16:
@Nobbyhexer Glad to hear that it works for you.

@Guilouz  I am in the process of testing it. What I noticed is that when I use this command `cd && wget https://github.com/Guilouz/Creality-K1-and-K1-Max/raw/main/Scripts/installer.sh` to download the script, it no longer creates a folder Creality und files for the sample configs. Is that correct?

Nastras, 2023-09-16:
Crap I saw it loads the required files via wget. Very well solved. I continue to test 😅

Nastras, 2023-09-16:
Ingenious. Now everything works as it should 💪
Can Moonraker be updated via the update manager in the future or is it necessary to do it via the script?

I have also tested the backup function and this also works. Perfect!

Next up is the Moonraker extension.

[image]

Guilouz, 2023-09-16:
> @Nobbyhexer Glad to hear that it works for you.
>
> @Guilouz I am in the process of testing it. What I noticed is that when I use this command `cd && wget [...]

Yes all is online now, it's better for updates.

closed 2023-09-16 by Guilouz