Guilouz/Creality-Helper-Script-Wiki #15: Configuration Mobilereaker

Nastras, 2023-09-16:
I need help again. I am in the process of testing Mobilereaker. I have installed the companion with the script and downloaded the app from the iOS appstore. Now it hangs when setting up in the app. I can't find good documentation either.
When I select simple and under printer address enter the ip of the printer 192.168.178.140 or 192.168.178.140:4409 I get an error on test connection.

An API key for Moonraker do I need this?

I hope I am on the right track?

[image]
[image]

Guilouz, 2023-09-16:
There is issue in simple mode, you must configure it in advanced mode.

Nastras, 2023-09-16:
Ohhh man, I couldn't have known that. I have now selected the advanced mode and here also nothing works I think it is because of my settings. Do you have a tip again? 🥹

[image]
[image]

Guilouz, 2023-09-16:
You have missed port in your ip and websocket.

Nastras, 2023-09-16:
Now I was able to establish a connection. Thanks for the hint. I had always tried with port but did not enter it at wegsacket.
A small error I still have with the cam and do not know exactly what mobileraker wants here?

[image]

Nastras, 2023-09-16:
I found this but I am not sure if it is the same error. I have reported this error there in the issue. Is the camera working for you?

https://github.com/Clon1998/mobileraker/issues/203

Guilouz, 2023-09-16:
Yes with this settings:

[image]
[image]

Nastras, 2023-09-16:
I can not select the camera function in the iOS app, I'll install the app again and set up again to see if it works. I will report.

closed 2023-09-16 by Guilouz

Nastras, 2023-09-16:
The camera stream is now displayed. I had to rename the camera to Mainsail for this. I had a space in the name and mobileraker did not like it. Thanks for the support 💪