Guilouz/Creality-Helper-Script-Wiki #17: Klipper Adaptive Meshing & Purging??

AiramsRT, 2023-09-17:
I have installed the plugin from the installer, but now I don't know how to make it work. Should I follow any more steps? In the creator's github it indicates adding some lines in monraker.cfg and printer.cfg and that I don't see them in my configuration, Do I have to add them? Because when I send to print it doesn't make any adaptive mesh.

Thanks!

Guilouz, 2023-09-17:
See my wiki for that.

And please do not open an issue for questions. There is discussions for that.

Use this only for report issue.

closed 2023-09-17 by Guilouz

AiramsRT, 2023-09-17:
Sorry, I had opened the wiki but it only showed how to install with the installer, not all the code was there.

Guilouz, 2023-09-17:
Refresh your page, all is there :)

AiramsRT, 2023-09-17:
Yesss!!

Thanks!!!