Guilouz/Creality-Helper-Script-Wiki #19: Camera has odd issue, also NGINX BAD GATEWAY

grrminator, 2023-09-19:
I would like to preface I have klipperized printers before. And I have klipperized this one with firmware 2 weeks ago or so.

I did the new 1.3.2.1 update, and followed the wiki and now orca throws this error on upload:

Error uploading to print host:
HTTP 502: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.17.7</center>
</body>
</html>

and on occasion I see the bad gateway error in browser.
[](url)

And now my camera does this.
[image]

The camera will work briefly after restoring the firmware and re-installing fluidd. But it WILL resume doing that.

It was not showing any of this prior.

Please assist.

Guilouz, 2023-09-19:
What settings do you use for camera ?

grrminator, 2023-09-19:
Default settings as in your guide.

grrminator, 2023-09-19:
Again, creality print webcam connection works. In Fluidd, it does NOT.

Guilouz, 2023-09-19:
So you have something wrong, camera working without issue in fluidd or mainsail and timelapse working too.

grrminator, 2023-09-19:
But I dont.

Its set up the exact same way on previous rooted firmwares.

I used your install script to install that stuff on there. Thats the ONLY
thing that has changed from previous firmware updates.

I have followed your guides before without issues, and others on the
subreddit. This problem ONLY occurs when I use your install script.

grrminator, 2023-09-19:
After some research, this is a KNOWN issue that can occur in
fluidd/mainsail.

Guilouz, 2023-09-19:
I can't reproduce this issue. Before release my script, i have tested it in all direction and removed and reinstall many times moonraker, fluidd or mainsail without issue.