Guilouz/Creality-Helper-Script-Wiki #20: Adaptive Meshing and Purging don't work

arlindonatal, 2023-09-19:
After installing and configuring KAMP, when printing, Meshing is done completely, not just in the printing area. Purging is done away from the printing area, always in the upper corner of the table.

I tried modifying START_PRINT to only do BED_MESH_CALIBRATE and LINE_PURGE, just like in the KAMP doc, but the same problem occurs.

**START_PRINT**
`[gcode_macro START_PRINT]
variable_prepare: 0
gcode:
  WAIT_TEMP_END
  CLEAR_PAUSE
  {% set g28_extruder_temp = printer.custom_macro.g28_ext_temp %}
  {% set bed_temp = printer.custom_macro.default_bed_temp %}
  {% set extruder_temp = printer.custom_macro.default_extruder_temp %}

  {% if 'BED_TEMP' in params|upper and (params.BED_TEMP|float) %}
    {% set bed_temp = params.BED_TEMP %}
  {% endif %}

  {% if 'EXTRUDER_TEMP' in params|upper and (params.EXTRUDER_TEMP|float) %}
    {% set extruder_temp = params.EXTRUDER_TEMP %}
  {% endif %}

  {% if printer['gcode_macro START_PRINT'].prepare|int == 0 %}
    {action_respond_info("not prepare.\n")}
    PRINT_PREPARE_CLEAR
    CX_ROUGH_G28 EXTRUDER_TEMP={extruder_temp} BED_TEMP={bed_temp}
    #CX_NOZZLE_CLEAR
    ACCURATE_G28
    #CX_PRINT_LEVELING_CALIBRATION
    BED_MESH_CLEAR
    BED_MESH_CALIBRATE
  {% else %}
    PRINT_PREPARE_CLEAR
  {% endif %}

  CX_PRINT_DRAW_ONE_LINE`

**CUSTOM MACRO**

`# Support for 1-wire based temperature sensors
#
# Copyright (C) 2020 Alan Lord <alanslists@gmail.com>
#
# This file may be distributed under the terms of the GNU GPLv3 license.
import time

class CUSTOM_MACRO:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.gcode = self.printer.lookup_object('gcode')
        self.pheaters = None
        self.heater_hot = None
        self.extruder_temp=None
        self.bed_temp=None
        self.prtouch = None
        #self.gcode.register_command("CX_PRINT_LEVELING_CALIBRATION", self.cmd_CX_PRINT_LEVELING_CALIBRATION, desc=self.cmd_CX_PRINT_LEVELING_CALIBRATION_help)
        self.gcode.register_command("CX_CLEAN_CALIBRATION_FLAGS", self.cmd_CX_CLEAN_CALIBRATION_FLAGS, desc=self.cmd_CX_CLEAN_CALIBRATION_FLAGS_help)
        self.gcode.register_command("CX_PRINT_DRAW_ONE_LINE", self.cmd_CX_PRINT_DRAW_ONE_LINE, desc=self.cmd_CX_PRINT_DRAW_ONE_LINE_help)
        self.gcode.register_command("CX_ROUGH_G28", self.cmd_CX_ROUGH_G28, desc=self.cmd_CX_ROUGH_G28_help)
        #self.gcode.register_command("CX_NOZZLE_CLEAR", self.cmd_CX_NOZZLE_CLEAR, desc=self.cmd_CX_NOZZLE_CLEAR_help)
        self.default_extruder_temp = config.getfloat("default_extruder_temp", default=240.0)
        self.default_bed_temp = config.getfloat("default_bed_temp", default=50.0)
        self.g28_ext_temp = config.getfloat("g28_ext_temp", default=140.0)
        self.nozzle_clear = config.getboolean('nozzle_clear', True)
        self.calibration = config.getint('calibration', default=0)
        self.leveling_calibration = 0
        pass

    def get_status(self, eventtime):
        return {
            'leveling_calibration': self.leveling_calibration,
            'default_extruder_temp': self.default_extruder_temp,
            'default_bed_temp': self.default_bed_temp,
            'g28_ext_temp': self.g28_ext_temp
        }

    cmd_CX_PRINT_LEVELING_CALIBRATION_help = "Start Print function,three parameter:EXTRUDER_TEMP(180-300),BED_TEMP(0-100),CALIBRATION(0 or 1)"
    def cmd_CX_PRINT_LEVELING_CALIBRATION(self, gcmd):
        self.gcode.run_script_from_command('CHECK_BED_MESH AUTO_G29=1')
        pass

    cmd_CX_CLEAN_CALIBRATION_FLAGS_help = "Clean calibration flags"
    def cmd_CX_CLEAN_CALIBRATION_FLAGS(self, gcmd):
        self.leveling_calibration = 0
        pass

    cmd_CX_PRINT_DRAW_ONE_LINE_help = "Draw one line before printing"
    def cmd_CX_PRINT_DRAW_ONE_LINE(self, gcmd):
      self.gcode.run_script_from_command('M83')
      self.gcode.run_script_from_command('SMART_PARK')
      self.gcode.run_script_from_command('G91')
      self.gcode.run_script_from_command('G1 Y-10 F600')
      self.gcode.run_script_from_command('G90')
      self.gcode.run_script_from_command('G1 Z0.8 F600')
      self.pheaters = self.printer.lookup_object('heaters')
      self.heater_hot = self.printer.lookup_object('extruder').heater
      self.gcode.respond_info("can_break_flag = %d" % (self.pheaters.can_break_flag))
      self.gcode.run_script_from_command('M104 S%d' % (self.extruder_temp))
      self.gcode.run_script_from_command('M140 S%d' % (self.bed_temp))
      self.pheaters.set_temperature(self.heater_hot, self.extruder_temp, True)
      self.gcode.respond_info("can_break_flag = %d" % (self.pheaters.can_break_flag))
      while self.pheaters.can_break_flag == 1:
         time.sleep(1)
      self.gcode.respond_info("can_break_flag = %d" % (self.pheaters.can_break_flag))
      if self.pheaters.can_break_flag == 3:
          self.pheaters.can_break_flag = 0
          self.gcode.respond_info("can_break_flag is 3")
          self.gcode.run_script_from_command('M220 S100')
          self.gcode.run_script_from_command('M221 S100')
          self.gcode.run_script_from_command('LINE_PURGE')
      pass    

    cmd_CX_ROUGH_G28_help = "rough G28"
    def cmd_CX_ROUGH_G28(self, gcmd):
        self.extruder_temp = gcmd.get_float('EXTRUDER_TEMP', default=self.default_extruder_temp, minval=180.0, maxval=320.0)
        self.g28_ext_temp = self.extruder_temp - 70
        if self.g28_ext_temp > 180.0:
            self.g28_ext_temp = 180.0
        try:
            self.prtouch = self.printer.lookup_object('prtouch_v2')
        except:
            self.prtouch = self.printer.lookup_object('prtouch')
            gcmd.respond_info("self.prtouch = prtouch")
        self.prtouch.change_hot_min_temp(self.g28_ext_temp)
        self.bed_temp = gcmd.get_float('BED_TEMP', default=self.default_bed_temp, minval=0.0, maxval=130.0)
        self.leveling_calibration = gcmd.get_int('LEVELING_CALIBRATION', default=1, minval=0, maxval=1)
        self.gcode.run_script_from_command('M104 S%d' % (self.g28_ext_temp))
        self.gcode.run_script_from_command('M140 S%d' % (self.bed_temp))
        self.gcode.run_script_from_command('M204 S500')
        self.gcode.run_script_from_command('G1 F2400 E-0.5')
        self.gcode.run_script_from_command('G28')
        # self.gcode.run_script_from_command('NOZZLE_CLEAR HOT_MIN_TEMP=%d HOT_MAX_TEMP=%d BED_MAX_TEMP=%d' % (self.g28_ext_temp, self.extruder_temp - 20, self.bed_temp))
        pass

    cmd_CX_NOZZLE_CLEAR_help = "nozzle clear with temperature"
    def cmd_CX_NOZZLE_CLEAR(self, gcmd):
        self.gcode.run_script_from_command('NOZZLE_CLEAR HOT_MIN_TEMP=%d HOT_MAX_TEMP=%d BED_MAX_TEMP=%d' % (self.g28_ext_temp, self.extruder_temp - 20, self.bed_temp))
        pass

def load_config(config):
    return CUSTOM_MACRO(config)
`

Guilouz, 2023-09-19:
Are you sure your gcode have label objects enabled ? Can you send me your gcode.

arlindonatal, 2023-09-19:
> Are you sure your gcode have label objects enabled ? Can you send me your gcode.

Yes, there is... 

[image]

**GCODE**
[Shape-Box.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/12663310/Shape-Box.zip)

luizfpgomes, 2023-09-19:
Yes, followed every step from kamp page, label object is active and Everything. Still does the line purge im the wrong placenta and does the full bed calibration. In klipper console it shows right at the start a error "unknown command "DEFINE_OBJECT" or something like that. I think the issue is in klipper version...but it's weird that yours it's working and ours don't hahaha

luizfpgomes, 2023-09-19:
[inomeado.pdf](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/12664145/inomeado.pdf)

here are the screenshots, let me know if you can find any issues in it...

Guilouz, 2023-09-19:
I see your gcode use DEFINE_OBJECT function, it's generated by PrusaSlicer:

DEFINE_OBJECT NAME=Shape_Box_id_0_copy_0 CENTER=110.000,109.985 POLYGON=[[100.21,100.21],[100.21,119.79],[119.79,119.79],[119.79,100.21]]

START_CURRENT_OBJECT NAME=Shape_Box_id_0_copy_0

My generated gcode by Orcaslicer use:

EXCLUDE_OBJECT_DEFINE NAME=OrcaCube_id_0_copy_0 CENTER=150,150 POLYGON=[[135,135],[160.981,135],[165,135],[165,160.981],[165,165],[139.019,165],[135,165],[135,139.019],[135,135]]

EXCLUDE_OBJECT_START NAME=OrcaCube_id_0_copy_0

Do you gave enabled enable_object_processing: True in mooraker.conf ?

Your issue is on slicer side.

luizfpgomes, 2023-09-19:
That was it, I changed to orca and everything worked fine! Thanks a lot for the help. It would be good to add a note about it in the change log.

Guilouz, 2023-09-19:
PrusaSlicer it supposed to work, I don't use it so no idea.

closed 2023-09-19 by Guilouz