Guilouz/Creality-Helper-Script-Wiki #23: KAMP Issue

VinoY27, 2023-09-21:
Hi Guilouz,
 Thank you so much for your effort on helping all of us to customize the K1, I was trying to setup KAMP in my K1, but i am running in to an issue, I have followed your steps. but not sure I am seeing error in console after the printer has completed the bed mesh, I have attached the console logs, gcode file used, klipper logs, moonraker logs and the config changes I have done in the  below Github link, please check when you find time. appreciate your help. please let me know if you need more info.

Error Screenshot:
[image]

KAMP_Error_Logs:
https://github.com/VinoY27/K1Configs/tree/main/KAMP_Error

Config I am using:
https://github.com/VinoY27/K1Configs/tree/main

I have updated the custom_macro.py has in the steps as well.
https://github.com/VinoY27/K1Configs/blob/main/custom_macro.py

I have unchecked the ARC settings in the slicer.
[image]

Thanks,
Vinoth

RandieBarsteward, 2023-09-21:
Same issue on larger prints but only when starting the print from the Fluidd interface, they seem fine when started via the Creality web interface.

grrminator, 2023-09-21:
Error persists after full KAMP guide.

grrminator, 2023-09-21:
11:59:18 
!! {"code":"key60", "msg":"Internal error on command:_BED_MESH_CALIBRATE", "values": ["_BED_MESH_CALIBRATE"]}
11:59:19 
!! {"code":"key60", "msg":"Internal error on command:BED_MESH_CALIBRATE", "values": ["BED_MESH_CALIBRATE"]}
11:59:19 
!! {"code":"key60", "msg":"Internal error on command:START_PRINT", "values": ["START_PRINT"]}

Creality print/fluidd now produce this isse.

VinoY27, 2023-09-21:
When starting the Print from the screen with "Print calibration" checked, KAMP mesh is not done, directly goes to print, please see the console logs below. 
[image]

Japhon128, 2023-09-21:
For what its worth, I was having a similar issue printing a 6x3 Gridfinity base on my K1 Max. It seemed like I was getting too close to the edges of the bed so I did a few things, although I'm not sure in what order or what actually did it. 
1. In KAMP_settings.cfg: I returned the KAMP margins back to default 
2. In printer.cfg: I returned mesh_min and mesh_max under [bed_mesh] back to default 
3. I rotated the model on Z-axis 45 degrees so it was not too close to the edges.
That seemed to work for me. Hopefully that help narrow things down a bit. Best of luck!

grrminator, 2023-09-21:
Also, if it does work, the full KAMP guide you posted causes the printer to do a bed mesh, then it does the bed mesh AGAIN but it will do one point, then go back to center, then another point, and back to center.

Which is frustrating because it makes the pre print time damn near 15 minutes.

talentedmrripley, 2023-09-22:
I too just experienced this failure.  If more logs or example gcode is needed I am happy to share it.

grrminator, 2023-09-25:
No fix I take it?

grrminator, 2023-09-25:
@Guilouz

Tombraider2006, 2023-09-26:
Unfortunately there will be a bug with the adaptive mesh. this is due to the fact that the prtuch module does not accept values other than 6 by 6, while your adaptive grid looks like a square it will work, but if the values are 6 by 4, 6 by 3 it will throw an error. So either place the model square or do not use the adaptive mesh

PA3GMT, 2023-10-03:
I also had issues with an non-square mesh (long in X-direction and narrow in Y-direction part on the buildplate) and solved it by adding a few extra lines to /usr/data/printer_data/config/KAMP/Adaptive_Meshing.cfg
Until Creality accepts non-square meshes you can use this as a workaround.
[Adaptive_Meshing.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/12796014/Adaptive_Meshing.zip)
I'm not a coding expert, there might be more elegant solutions but it does the job.

Guilouz, 2023-10-03:
> I also had issues with an non-square mesh (long in X-direction and narrow in Y-direction part on the buildplate) and solved it by adding a few extra lines to [...]

You confirmed just adding this fix the issue ? :

########################################
    {% if (points_x > points_y) %}
        {% set points_y = points_x %}
    {% endif %}

    {% if (points_x < points_y) %}
        {% set points_x = points_y %}
    {% endif %}
	#######################################

PA3GMT, 2023-10-03:
Yes, I confirm that you have to add these lines.

Guilouz, 2023-10-03:
But that means mesh is bigger than print no ?

PA3GMT, 2023-10-03:
The mesh is not square but there are an equal number of mesh points in X
and Y direction.

Guilouz, 2023-10-03:
Thanks for this, i will update script for that.

closed 2023-10-03 by Guilouz

talentedmrripley, 2023-10-04:
I have also tested the change @PA3GMT provided and can confirm that it seem to resolve the issue for me as well.

Guilouz, 2023-10-23:
> Doesn't fix the print issue. Just trying to print a 220x220 bed level test causes the same KAMP error. Disabling KAMP, it prints fine. Anything above 210 wide it comes with the same error.

Can you try this files by replacing by existing one :
 
[New KAMP.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13077998/New.KAMP.zip)

DjFillament, 2023-10-24:
> Can you try this files by replacing by existing one :
>
> [New KAMP.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13077998/New.KAMP.zip)

Dang no still freezing when it parks in corner and errors out.

jwcoope, 2023-11-20:
Need to change the pr_version to 1 instead of 2.  Looks like version 2 has a bug that you can't go above 36 probe points.  Changing the setting below fixes this.  This is not an issue with KAMP or installation script it's a bug with creality. 

printer.cfg
[prtouch_v2]
pr_version: 1