Guilouz/Creality-Helper-Script-Wiki #24: Bad Gateway 502 - Moonranker (tmp)

thebayer, 2023-09-21:
I am facing problems with sending large files through Moonranker on my K1 Max
It fills the tmp partition and gives the Bad Gateway 502 error
[image]
97.4M Sep 21 09:40 moonraker.upload-49100.mru

Then I need to delete the generated file (e.g. moonraker.upload-49100.mru), restart the services and send files smaller than 70mb

Guilouz, 2023-09-21:
Max upload file size configured in Moonraker is 1024 MiB.

thebayer, 2023-09-21:
Many thanks for the reply.
but /tmp has a maximum size of 100mb.
when the file is larger than the free space, it gives the Bad Gateway error...
;-(

thebayer, 2023-09-21:
Many thanks for the reply.
but /tmp has a maximum size of 100mb.
When the file is larger than the free space in /tmp, it gives the Bad Gateway error...
;-(

Guilouz, 2023-09-21:
Can you send me your gcode file to try when i go home.

thebayer, 2023-09-21:
Just generate any Gcode larger than 100MB and send it via Moonranker (Orca, Prusa).
Below is the link to a GCode https://we.tl/t-UfrGNBv0TN
this file can be found on Thingverse: https://www.thingiverse.com/thing:4543820

Guilouz, 2023-09-22:
It's now fixed. Please remove moonraker and reinstall it from my script.

closed 2023-09-22 by Guilouz

thebayer, 2023-09-23:
You are THE GUY !!!!!! TKS !!!!!!!!