Guilouz/Creality-Helper-Script-Wiki #25: update_manager mobileraker

Nastras, 2023-09-22:
Hello @Guilouz 

I have tried to customize the update manager for mobileraker companion. But I get errors there with this entry. Would it not be useful if the script directly enters the correct update path in monraker.cfg during installation?

[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
virtualenv: ~/mobileraker-env
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh

https://github.com/Clon1998/mobileraker_companion

[image]

joschiv1977, 2023-09-22:
For mobileraker_companion there is no real benefit. and if you look into the scripts there are changes for the K1 there are not included into the original repo. So if you use the original mobileraker companion it will break it on your rooted K1.

Nastras, 2023-09-22:
Thank you for the explanation. I was not aware of that. Then we can close here.

closed 2023-09-22 by Nastras