Guilouz/Creality-Helper-Script-Wiki #26: Uploading large gcode files fails

talentedmrripley, 2023-09-24:
I attempted to upload a gcode file of 108MBs.  The upload reached about 95% and then failed.  I retried several times but all failed.  I was then unable to upload any gcode of any size, even less than 1MB.  The moonracker.log showed the following:

`2023-09-23 22:01:09,635 [app.py:prepare()] - Upload Request Received from 172.16.10.109
2023-09-23 22:01:09,668 [http1connection.py:__exit__()] - Uncaught exception
Traceback (most recent call last):
  File "/usr/data/moonraker/moonraker-env/lib/python3.8/site-packages/tornado/http1connection.py", line 661, in _read_fixed_body
    await ret
  File "/usr/data/moonraker/moonraker/moonraker/app.py", line 908, in data_received
    await evt_loop.run_in_thread(self._parser.data_received, chunk)
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/lib/python3.8/site-packages/streaming_form_data/parser.py", line 60, in data_received
    result = self._parser.data_received(data)
  File "streaming_form_data/_parser.pyx", line 212, in streaming_form_data._parser._Parser.data_received
  File "streaming_form_data/_parser.pyx", line 401, in streaming_form_data._parser._Parser._parse
  File "streaming_form_data/_parser.pyx", line 398, in streaming_form_data._parser._Parser._parse
  File "streaming_form_data/_parser.pyx", line 190, in streaming_form_data._parser._Parser.on_body
  File "streaming_form_data/_parser.pyx", line 117, in streaming_form_data._parser.Part.data_received
  File "/usr/lib/python3.8/site-packages/streaming_form_data/targets.py", line 41, in data_received
    self.on_data_received(chunk)
  File "/usr/lib/python3.8/site-packages/streaming_form_data/targets.py", line 103, in on_data_received
    self._fd.write(chunk)
OSError: [Errno 28] No space left on device`

So I started poking around in the file system:

`root@creality /tmp [#] df -h
Filesystem                Size      Used Available Use% Mounted on

/dev/root               107.9M    107.9M         0 100% /rom
devtmpfs                 98.4M         0     98.4M   0% /dev
tmpfs                    98.5M     16.0K     98.5M   0% /dev/shm
tmpfs                    98.5M     98.5M         0 100% /tmp
tmpfs                    98.5M    252.0K     98.3M   0% /run
/dev/mmcblk0p9           96.8M      2.1M     89.7M   2% /overlay
overlayfs:/overlay       96.8M      2.1M     89.7M   2% /
/dev/mmcblk0p10           6.5G      1.1G      5.1G  18% /usr/data`

Looking in /tmp:

`root@creality /tmp [#] ls -lah
total 98M
  560 Sep 23 22:21 .
 1.0K Sep 17 13:21 ..
  231 Mar  1  2020 .cam_version
  237 Mar  1  2020 .mcu_version
    0 Mar  1  2020 ai_server_uds
    0 Mar  1  2020 bsa_server.log
    0 Mar  1  2020 camera_main
  120 Mar  1  2020 creality
   60 Mar  1  2020 dbus
  100 Mar  1  2020 hci_snoop.log
   10 Sep 23 22:16 klipper_host_mcu -> /dev/pts/0
    0 Sep 23 22:16 klippy_client
    0 Sep 23 22:16 klippy_uds
    0 Sep 23 21:45 load_done
  744 Mar  1  2020 mcu_update.log
49.5K Sep 23 22:21 messages
98.3M Sep 23 22:18 moonraker.upload-1984.mru
  240 Mar  1  2020 nginx
   10 Sep 23 22:16 printer -> /dev/pts/1
   94 Mar  1  2020 resolv.conf
   60 Mar  1  2020 subsys
    0 Mar  1  2020 sys_sock
    0 Mar  1  2020 uvc_fifo
    0 Mar  1  2020 wpa_ctrl_1446-0
    0 Mar  1  2020 wpa_ctrl_1446-1
    0 Mar  1  2020 wpa_ctrl_1446-2
    0 Sep 23 21:45 wpa_ctrl_1941-0
    0 Sep 23 21:45 wpa_ctrl_1946-0`

I saw the file moonraker.upload-1984.mru and a recent time stamp.  Deleting this file allowed me to upload other smaller files.  But it looks like the way moonraker handles uploads is to write them to /tmp first before moving them to /usr/data/printer_data/gcodes and the total space of /tmp is 98.5MB so anything larger than that will fail.

Nastras, 2023-09-24:
Have you reinstalled moonraker yet?

https://github.com/Guilouz/Creality-K1-and-K1-Max/issues/24

Guilouz, 2023-09-24:
This is already fixed. See changelog.

closed 2023-09-24 by Guilouz