Guilouz/Creality-Helper-Script-Wiki #27: Feature Request for Input Shaper results as a graphic

Nastras, 2023-09-25:
Hello

Would it be possible for you to make an option available in your script so that the results of the input shaper are output as a graphic?

I have often seen pictures in the community where the results of the Input Shaper were graphically displayed. While looking for how this works I found this script in the Klipper repo.

https://github.com/Klipper3d/klipper/blob/master/scripts/graph_shaper.py

I have to admit that I don't know how to use it.

If you can, I think it would be a great addition.

Thank you!
Best regards

Nastras, 2023-09-25:
[image]

joschiv1977, 2023-09-25:
Out of the box (root) the script is not working:

/usr/share/klippy-env/bin/python /usr/share/klipper/scr
ipts/calibrate_shaper.py  /tmp/calibration_data_x_20230924_212842.csv  -o /tmp/
shaper_calibrate_x.png
Traceback (most recent call last):
  File "/usr/share/klipper/scripts/calibrate_shaper.py", line 174, in <module>
    main()
  File "/usr/share/klipper/scripts/calibrate_shaper.py", line 156, in main
    selected_shaper, shapers, calibration_data = calibrate_shaper(
  File "/usr/share/klipper/scripts/calibrate_shaper.py", line 44, in calibrate_shaper
    helper = shaper_calibrate.ShaperCalibrate(printer=None)
  File "/usr/share/klipper/scripts/../klippy/extras/shaper_calibrate.py", line 88, in __init__
    configfile = self.printer.lookup_object('configfile')
AttributeError: 'NoneType' object has no attribute 'lookup_object'

Guilouz, 2023-09-25:
Yes, the environments having been modified on the K1 OS, it's necessary to convert the CSVs from another machine that supports python.

closed 2023-09-25 by Guilouz