Guilouz/Creality-Helper-Script-Wiki #28: shaper calibrate

Tombraider2006, 2023-09-26:
the native script shaper sucks, why repeat the mistakes of underprogrammers? Firstly, stopping the sensors is not necessary at all. This is the delirium of an inflamed Chinese brain. the second calculation of coordinates and moving to the test location does not make sense since the test coordinates are registered in the resonance_test section, doing the test separately also makes little sense, so the inputshaper section:

Guilouz, 2023-09-26:
SHAPER_CALIBRATE Command do input shaper on x axis first and then on y axis.

And this is not what we want here because the test result of the y axis replace that of the x axis. the test must be carried out separately for each x starting with y.

closed 2023-09-26 by Guilouz

Tombraider2006, 2023-09-26:
at least try to do it. the answer will surprise you. I know about the replacement. but my script works and has been tested by a whole group in telegram. More than 300 printers work with this script and the result is not the same on the axes.

Guilouz, 2023-09-26:
Yes it's working now. I think they have fixed this because it was not possible before.

Tombraider2006, 2023-09-26:
> Yes it's working now. I think they have fixed this because it was not possible before.

unfortunately they didn't fix anything. does not work correctly when specifying an axis argument. if applied without an argument it works starting from 1.2.9.12