Guilouz/Creality-Helper-Script-Wiki #29: TLS Error

Saberlord, 2023-09-26:
Currently I am trying to update with the installer, but every wget request made, will always return with a TLS certificate validation error

The running code:

root@creality /root [#] cd && wget https://github.com/Guilouz/Creality-K1-and-K1-Max/raw/main/Scripts/installer.sh
Connecting to github.com (20.205.243.166:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer

It runs through the whole script as well.

Guilouz, 2023-09-26:
Yes it's normal it's related to OS but there is no issue.

closed 2023-09-26 by Guilouz

Saberlord, 2023-09-27:
In that case, how do I avoid getting that error? Change the OS? I used to update with installation script 1.3, but I can't now.

I have also removed Moonraker as I thought that was the issue and now I can't use my printer

Saberlord, 2023-09-27:
That is the problem, with the TLS error, the SSH cannot get any response from the server so the script cannot pull any files from github.*
[image]
**

This is what i get when i run the installer script that i transferred into the printer

Guilouz, 2023-09-27:
Working fine for me :

[image]

Guilouz, 2023-09-27:
Just updated my script by removing TLS error.

Saberlord, 2023-09-28:
I just tried, seems like it is not the script issue. 

It seems like my board has some issues connecting to web addresses. Found out that the main error is 
" TLS error from peer (alert code 80): 80" and
"error getting response: Connection reset by peer"

not sure what caused the printer to get bad ports from web addresses but seems fine when I try to ping IP addresses. 

Tried resetting the printer, restore old firmware and updated but still stuck.

Saberlord, 2023-09-28:
I have managed to split the install file and replaced the web pulls with direct transferring to the exact folder.

Thanks.

MuaiyadTSB, 2023-09-29:
Getting same error here when pulling the script, same error occurs as well across the helper script if I move it manually. 

Pulling other files from other repositories, works fine.

Pulling the helper script works fine on the same network if I attempt to do it on my computer, but fails when attempting  to do it from the K1.

Manually installing entware and updating busybox and wget-ssl seems to fix the issue, but breaks mainsail/fluidd for some reason.

Thanks for all of your work.

[image]

Guilouz, 2023-09-29:
Try with this command: 

```
  cd && wget --no-check-certificate https://github.com/Guilouz/Creality-K1-and-K1-Max/raw/main/Scripts/installer.sh
  ```

MuaiyadTSB, 2023-09-29:
I found the suggested command on multiple forums while trying troubleshoot, but it made no difference. Using raw github url seems to fix the issue for this repo, I created a pull request with the updated url in the script. It seems to resolve the issue for me.