Guilouz/Creality-Helper-Script-Wiki #30: some macro to add шт usefull macro

Tombraider2006, 2023-09-26:
[image]
```
[duplicate_pin_override]
pins: PC0, PC5

[temperature_fan chamber]
pin: PC0
cycle_time: 0.0100
hardware_pwm: false
max_power: 1
shutdown_speed: 0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
min_temp: 0
max_temp: 70
control: watermark
max_delta: 2
target_temp: 40.0
max_speed: 1.0
min_speed: 0.0

[gcode_macro M141]
description: Set temperature of chamber fan
gcode:
    SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=chamber TARGET={params.S|default(30)}
```
1. will add temperature settings in the printer chamber. also m141 to control it using commands from the slicer 

2. on motherboard have buzzer but it in pi part and for use it in config need to do a few non-obvious things 
login as root
```
wget -P /usr/share/klipper/klippy/extras/ https://raw.githubusercontent.com/Tombraider2006/K1/main/filament/gcode_shell_command.py
wget -P /usr/data/printer_data/config/ https://raw.githubusercontent.com/Tombraider2006/K1/main/filament/11.wav
```
add to printer.cfg or gcode_macro.cfg
```
[gcode_shell_command beep]
command: aplay /usr/data/printer_data/config/11.wav
timeout: 2.
verbose: False

[gcode_macro BEEP]
 gcode:
  RUN_SHELL_COMMAND CMD=beep
  RUN_SHELL_COMMAND CMD=beep
  RUN_SHELL_COMMAND CMD=beep
  RUN_SHELL_COMMAND CMD=beep
```
Now the BEEP macro can be used as you like. I added it to the end of the end_print macro so that after printing is completed it would call me. you can also add it to the config of the filament sensor so that when the filament breaks, it will give a sound signal.

I think you can find another way to avoid downloading the WAV file, but I don’t know that.

3. some fix macro:

This fix improves the hotend airflow at the end of the print and makes it quieter and faster by distributing the power of the side airflow and hotend airflow coolers

```
[gcode_macro WAIT_TEMP_START]
gcode:
  UPDATE_DELAYED_GCODE ID=wait_temp DURATION=1
  M106 P0 S200
  SET_PIN PIN=fan2 VALUE=80

[gcode_macro WAIT_TEMP_END]
gcode:
  UPDATE_DELAYED_GCODE ID=wait_temp DURATION=0
  M106 P0 S0
  M106 P2 S0
```

Guilouz, 2023-09-27:
1. will add temperature settings in the printer chamber. also m141 to control it using commands from the slicer

What is this supposed to do? Target does nothing and not control the back fan (fan1), just see it in graph.

Tombraider2006, 2023-09-28:
> What is this supposed to do? Target does nothing and not control the back fan (fan1), just see it in graph.
https://www.klipper3d.org/Config_Reference.html#temperature_fan 
read the following manual

Guilouz, 2023-09-28:
> > What is this supposed to do? Target does nothing and not control the back fan (fan1), just see it in graph.
> > https://www.klipper3d.org/Config_Reference.html#temperature_fan
> > read the following manual

Yes I know what this is supposed to do however, we agree that the fan is supposed to come on when the target temperature is reached, right ?

Currently my chamber reads 25 degrees, so if I tell it a target temperature of 20 degrees the fan is supposed to come on and that's not the case.

Tombraider2006, 2023-09-28:
[image]
ы
[image]

special for you. trying right now all work. are you copy all code?

```

[duplicate_pin_override]
pins: PC0, PC5

[temperature_fan chamber]
pin: PC0
cycle_time: 0.0100
hardware_pwm: false
max_power: 1
shutdown_speed: 0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
min_temp: 0
max_temp: 70
control: watermark
max_delta: 2
target_temp: 40.0
max_speed: 1.0
min_speed: 0.0

[gcode_macro M141]
description: Set temperature of chamber fan
gcode:
    SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=chamber TARGET={params.S|default(30)}
```
you need all parts to work propertly

Guilouz, 2023-09-28:
Yes but slider is not controlled, it's always stay at 0.

Tombraider2006, 2023-09-28:
> Yes but fan1 slider is not controlled, it's always stay at 0.
> 
> [image]

oh that's what you're talking about. well, OK. The thing is that this life hack is a bit cheating. If you do it normally, you need to remove the chamber temperature sensor and chamber cooler and use a temperature chamber fan instead, but this can ruin Chinese scripts. if you just add it, there will be a config error. Therefore, first we insert a section that allows duplicate pins and only then add our config.

so you won’t be able to see the fan work because another part of the config is working

Guilouz, 2023-09-28:
For buzzer support, i will add support in my script with next release.

Tombraider2006, 2023-09-28:
> For buzzer support, i will add support in my script with next release.

Maybe you can find a better way than me - without using a WAV file. I did not find.

Guilouz, 2023-09-28:
> > For buzzer support, i will add support in my script with next release.
> 
> Maybe you can find a better way than me - without using a WAV file. I did not find.

Yes it's possible if we found buzzer pin.

Nazgile94, 2023-09-28:
quick question Tombraider

> [gcode_macro WAIT_TEMP_START]
> gcode:
>   UPDATE_DELAYED_GCODE ID=wait_temp DURATION=1
>   M106 P0 S200
>   SET_PIN PIN=fan2 VALUE=80
> 
> [gcode_macro WAIT_TEMP_END]
> gcode:
>   UPDATE_DELAYED_GCODE ID=wait_temp DURATION=0
>   M106 P0 S0
>   M106 P2 S0

how do i implement this?
i added this to the macro conf.
did i need to implement this gcode in my slicer at the end of the print? or how do i exactly activate this after each print? - sry for asking, kind of newbie :)

Tombraider2006, 2023-09-28:
> i added this to the macro conf.

these are actually existing macros in the file gcode_macro.cfg. they need to be found and replaced with mine. at the end of the print the head cooler will blow a little quieter and the side one will blow quietly. Previously, only the head cooler was blowing at 100%

Guilouz, 2023-09-29:
All is done, see [Improve Fans Control](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Improve-Fans-Control)

closed 2023-09-29 by Guilouz