Guilouz/Creality-Helper-Script-Wiki #34: fix issue with unput sharper

Nazgile94, 2023-09-29:
https://github-wiki-see.page/m/Guilouz/Creality-K1-and-K1-Max/wiki/Fix-issue-with-Input-Shaper

on my k1 max 

i get everytime while run the macro:

crash after 30-40 seconds, high cpu load and

{"code": "", "msg":"MCU 'mcu' shutdown: Move queue overflow", "values": []}

so i think its to mutch for the little cpu? 🗡️ 

`10:40:26 
// Wait for calculations..
10:40:27 
// Wait for calculations..
10:40:30 
// Wait for calculations..
10:40:35 
// Wait for calculations..
10:40:40 
// Wait for calculations..
10:40:45 
// Wait for calculations..
10:40:50 
// Wait for calculations..
10:40:55 
// Wait for calculations..
10:41:00 
// Wait for calculations..
10:41:03 
// calc_freq_response return (0)
10:41:03 
// Testing axis y
10:41:06 
// Disabled [input_shaper] for resonance testing
10:41:06 
// Testing frequency 5 Hz
10:41:06 
// Testing frequency 6 Hz
10:41:06 
// Testing frequency 7 Hz
10:41:06 
// Testing frequency 8 Hz
10:41:06 
// Testing frequency 9 Hz
10:41:09 
// Testing frequency 10 Hz
10:41:10 
// Testing frequency 11 Hz
10:41:11 
// Testing frequency 12 Hz
10:41:12 
// Testing frequency 13 Hz
10:41:13 
// Testing frequency 14 Hz
10:41:14 
// Testing frequency 15 Hz
10:41:15 
// Testing frequency 16 Hz
10:41:16 
// Testing frequency 17 Hz
10:41:17 
// Testing frequency 18 Hz
10:41:18 
// Testing frequency 19 Hz
10:41:20 
// Testing frequency 20 Hz
10:41:20 
// Klipper state: Shutdown
10:41:21 
!! {"code":"key60", "msg":"Internal error on command:SHAPER_CALIBRATE", "values": ["SHAPER_CALIBRATE"]}
10:41:21 
!! {"code":"key60", "msg":"Internal error on command:INPUT_SHAPER", "values": ["INPUT_SHAPER"]}`

Guilouz, 2023-09-29:
did you change your microsteps?

Nazgile94, 2023-09-29:
ehmmm i dont know - i dont think so :D just gained root -> downloaded the script, installed fluidd :D

Nazgile94, 2023-09-29:
ok it crashs for the normal inputsharper test too - so maybe i did something wrong :O 

any idea what settings can cause this problem?

Guilouz, 2023-09-29:
It depends on what you really did

Nazgile94, 2023-09-29:
root->installed the helperscript -> installed moonraker+nginx, fluid, companion app, bed leveler - litteraly all of the script without mainsail.

changed gcode for the inputsharper adjust and clicked on fluidd the z-offset button bcs i thinked it was for lift up / down :D - but reversed it to zero and runned the auto leveling process of printer after that.

maybe i should reset factory settings?

Guilouz, 2023-09-29:
Try to factory reset, it's the first thing to do after update to be sure to have a clean install.

I have run SHAPER_CALIBRATE command 3 times in a row without any problem

Nazgile94, 2023-09-29:
i will do factory reset -> root-> install moonraker/nginx + fluidd with helperscript - and insert the gcode for the input sharper fix 

i will respond as soon im done with it ! :-)

Nazgile94, 2023-09-29:
okay, after factoryreset - it works!

dont know what i have done xD 

ty!!!

closed 2023-09-29 by Nazgile94