Guilouz/Creality-Helper-Script-Wiki #35: FEATURE REQUEST - OBICO

Nazgile94, 2023-09-29:
https://www.reddit.com/r/crealityk1/comments/16n1ip8/guide_get_obico_running_on_your_k1/

Maybe you can implement the installation of Obico in the script? 

tested it manually  on my k1 max - works! 

https://www.reddit.com/r/crealityk1/comments/16n1ip8/comment/k2kcpi6/?utm_source=share&utm_medium=web2x&context=3

Guilouz, 2023-09-30:
I don't use Obico so i can't try it.

If you give me all commands I can integrate it to the script.

Nazgile94, 2023-10-01:
#Entware required!#
# running obico server is required, like inside docker#

cd /usr/data
git clone https://github.com/TheSpaghettiDetective/moonraker-obico
cd moonraker-obico
cp moonraker-obico.cfg.sample /usr/data/moonraker-obico/moonraker-obico.cfg

(ask question if selfhosted or not - if not - skip this - if yes, change the "url = https://app.obico.io" to your own server url inside the moonraker-obico.cfg - like url = http://192.168.0.100:port)

change "logpath" inside the moonraker-obico.cfg  old: path = /home/pi/printer_data/config/moonraker-obico.log  new: "/usr/data/moonraker-obico/moonraker-obico.cfg"

cd /usr/bin
rm virtualenv
easy_install virtualenv

cd /usr/data/moonraker-obico
virtualenv env  
source env/bin/activate    (takes some time)

pip3 install --require-virtualenv requests==2.26.0

pip3 install --require-virtualenv backoff==1.11.1

pip3 install --require-virtualenv sentry-sdk==1.9.8

pip3 install --require-virtualenv bson==0.5.10

pip3 install --require-virtualenv pathvalidate==2.5.0

pip3 install --require-virtualenv sarge==0.1.6

pip3 install --require-virtualenv websocket-client==1.2.1

pip3 install --require-virtualenv ipdb==0.13.9

pip3 install --require-virtualenv flask==2.1.2

pip3 install --require-virtualenv netaddr==0.8.0

opkg install python3-psutil

cp -R /opt/lib/python3.11/site-packages/psutil* /usr/data/moonraker-obico/env/lib/python3.8/site-packages/

python3 -m moonraker_obico.link -c /usr/data/moonraker-obico/moonraker-obico.cfg  (insert obico link code)

find /usr/data/moonraker-obico/ -name "*.pyc" -delete

the start script is inside the /usr/data/moonraker-obico folder:

/usr/data/moonraker-obico/env/bin/python3 -B -m moonraker_obico.app -c /usr/data/moonraker-obico/moonraker-obico.cfg

maybe implement a init.d script to run this at boot.

Nazgile94, 2023-10-01:
example moonraker-obico.cfg

`[server]
url = https://app.obico.io
# auth_token: <let the link command set this, see more in readme>
# sentry_opt: out or in

[moonraker]
host = hostip - like printers ip:fluiddport
port = 7125
# api_key = <grab one or set trusted hosts in moonraker>

[webcam]
disable_video_streaming = False

# CAUTION: Don't modify the settings below unless you know what you are doing
#   In most cases webcam configuration will be automatically retrived from moonraker
#
# Lower target_fps if ffmpeg is using too much CPU. Capped at 25 for Pro users (including self-hosted) and 5 for Free users
# target_fps = 25
#
# snapshot_url = http://127.0.0.1:8080/?action=snapshot
# stream_url = http://127.0.0.1:8080/?action=stream
# flip_h = False
# flip_v = False
# rotate_90 = False
# aspect_ratio_169 = False

[logging]
path = /home/pi/printer_data/config/moonraker-obico.log
level = INFO

[tunnel]
# CAUTION: Don't modify the settings below unless you know what you are doing
# dest_host = 127.0.0.1
# dest_port = 80
# dest_is_ssl = False`

Guilouz, 2023-10-03:
Also needed script for removing obico, i see it touch to virtualenv.

TaylorN15, 2023-10-06:
[Here ](https://www.reddit.com/r/crealityk1/comments/16n1ip8/guide_get_obico_running_on_your_k1/) is a guide that worked for me, before I decided to use Octoeverywhere/Mobileraker instead

IriGitt, 2023-10-11:
> [Here ](https://www.reddit.com/r/crealityk1/comments/16n1ip8/guide_get_obico_running_on_your_k1/) is a guide that worked for me, before I decided to use Octoeverywhere/Mobileraker instead

Do you have a working install for Octoeverywhere? The one that I found doesnt work.

TaylorN15, 2023-10-11:
> Do you have a working install for Octoeverywhere? The one that I found doesnt work.

https://github.com/D3vil-Design/K1-OctoEverywhere

Guilouz, 2023-10-22:
https://www.obico.io/blog/remote-access-creality-k1/

closed 2023-10-22 by Guilouz