Guilouz/Creality-Helper-Script-Wiki #37: Feature Request: Fan rename with tachometer_pin

Nastras, 2023-09-29:
I have just found an improvement for the fan support at K1. Here the names are assigned correctly. It works great so far.

I would be happy if it finds its way into the script.
Source: https://github.com/meteyou/creality-k1/blob/main/klipper/config/addons/fans.cfg

[image]

`[duplicate_pin_override]
pins: PC6, nozzle_mcu:PB4, nozzle_mcu:PB3

[fan_feedback]
# fan0_pin: nozzle_mcu:PB3 # model fan
fan0_pin: nozzle_mcu:PB4 # throat fan
fan1_pin: PC6 # backplane fan
print_delay_time: 5
current_delay_time: 2

# this is needed for the part cooling fan
[static_digital_output my_fan_output_pins]
pins: nozzle_mcu: PB6

# part cooling fan
[fan]
pin: !nozzle_mcu: PB8
cycle_time: 0.0100
off_below: 0.1
tachometer_pin: nozzle_mcu:PB3

# big part cooling fan on the side
[fan_generic sideblower]
pin: PB1
cycle_time: 0.0100
kick_start_time: 0.5
off_below: 0.2

# fan in the bottom case
[controller_fan controller]
pin: PB2
shutdown_speed: 1
fan_speed: 1.0
idle_timeout: 30
idle_speed: 0.8
heater: extruder, heater_bed
tachometer_pin: PC6

# coldend fan
[heater_fan coldend]
pin: nozzle_mcu:PB5
heater: extruder
heater_temp: 40
tachometer_pin: nozzle_mcu:PB4

# gcode_macro from creality to control all fans. this is modified to use the klipper fan options
[gcode_macro M106]
rename_existing: M106.1
gcode:
  {% set fans = printer["gcode_macro PRINTER_PARAM"].fans|int %}
  {% set fan = 0 %}
  {% set value = 0 %}
  {% if params.P is defined %}
  {% set tmp = params.P|int %}
  {% if tmp < fans %}
  {% set fan = tmp %}
  {% endif %}
  {% endif %}
  {% if params.S is defined %}
  {% set tmp = params.S|float %}
  {% else %}
  {% set tmp = 255 %}
  {% endif %}
  {% if tmp > 0 %}
  {% if fan == 0 %}
  {% set value = tmp %}
  {% else %}
  {% set value = tmp / 255 %}
  {% endif %}
  {% endif %}
  {% if fan == 0 %}
  M106.1 S{value}
  {% elif fan == 2 %}
  SET_FAN_SPEED FAN=sideblower SPEED={value}
  {% endif %}

[gcode_macro M107]
rename_existing: M107.1
gcode:
  {% if params.P is defined %}
  {% set fan = params.P|int %}
  {% if fan == 0 %}
  M106 P0 S0
  {% elif fan == 2 %}
  SET_FAN_SPEED FAN=sideblower SPEED=0
  {% endif %}
  {% else %}
  M106 P0 S0
  SET_FAN_SPEED FAN=sideblower SPEED=0
  {% endif %} `

Guilouz, 2023-09-29:
This can't be done by script, it's just changes in config files.

closed 2023-09-29 by Guilouz

Nastras, 2023-09-29:
Thanks for the quick feedback. Maybe it would be something for the new section improve fans control.

Guilouz, 2023-09-30:
And this is not updated, it's an old config.

Nastras, 2023-09-30:
Hmmm, that's a good point. I had not been so far in testing that I noticed that certain macros no longer work or dependencies exist there.

Unfortunately, I'm not a programmer find your project for the K1 but totally great. Therefore, I apologize that I might be a little too euphoric here suggestions for improvement. I just want to be helpful and bring the project further 😅

Best regards
Nastras