Guilouz/Creality-Helper-Script-Wiki #38: Feature Request: Fluidd & Mainsail K1 Themes

Nastras, 2023-09-29:
Cool design for the GUI:
https://github.com/tlace17/Creality_Mainsail_Themes
https://github.com/tlace17/Creality_Fluidd_Themes

Guilouz, 2023-09-29:
His fluidd theme is not very well designed because it does not allow you to choose the color with the color picker, mine does:

[image]

[image]

[image]

tlace17, 2023-09-30:
> His fluidd theme is not very well designed because it does not allow you to choose the color with the color picker, mine does:
> 
> [image] [image] [image]

I agree, for one reason or another I was having issues figuring out how that functionality works. Nice work getting it to work, I am learning as I go and I saw that nobody had made one yet. What did you have to do to get that to work? My Mainsail themes are a bit more involved but I could probably make them better.

Guilouz, 2023-09-30:
SVG file must be have no infill and point out to the correct fluidd class.

I will share my theme.

closed 2023-09-30 by Guilouz

tlace17, 2023-09-30:
> SVG file must be have no infill and point out to the correct fluidd class.
> 
> I will share my theme.

Thanks! Mine didn't have any infill. I will have to check out what you mean by Fluidd class.

Guilouz, 2023-09-30:
I'm working on this theme:

[image]

[image]

Nastras, 2023-11-11:
@Guilouz What has actually happened to the topic here? Neither solution has made it into the script yet ☹️.