Guilouz/Creality-Helper-Script-Wiki #40: white and black timelapse

le-pet, 2023-10-01:
all is in the title, the timelapse encoding is in white and black, i don't know if it's the setting yuv420p or not

Guilouz, 2023-10-01:
I have tested them in fluidd and it's working fine for me.``

Make sure you have latest timelapse version i have updated few days ago.

BrianP525, 2023-10-01:
I am also having this issue. Just performed full install today using the tools provided. (Awesome project btw)  If I export the images, the zip contains all the pictures and they look fine. When it renders, there is a green bar at the bottom right, the colors appear off, looks mostly black and white, but the yellow sticker on the printhead appears as black and white, but you can also see an artifact of this label appear green and squished, but in 2 other areas of the image, one to the left, and another to the right.

[image]

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/146669611/22acfdcc-596b-498a-b308-9b1a8fb6c017

le-pet, 2023-10-01:
tested in fluidd and moonraker, same, as described by @BrianP525, updated automatically the script, and in case of manually to, reboted the printer, same, the problem is about 2 of my k1

Guilouz, 2023-10-01:
Can you try this file after deleting old timelapse.py and timelapse.pyc files from /usr/data/moonraker/moonraker/moonraker/components/

[timelapse.py.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/12778377/timelapse.py.zip)

BrianP525, 2023-10-02:
Deleted the 2 timelapse.py and .pyc files, uploaded the timelapse.py from the zip, restarted and tested. Still have the same result, pictures are fine, video render is bad. I also found that there is a separate timelapse that records in the creality side of things, that seems to be only accessible through creality print, and that timelapse looks fine.
[image]

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/146669611/f9e8811c-9e8e-453f-88ab-b67052cff46d

Guilouz, 2023-10-02:
Can you try this one:

[timelapse.py.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/12781383/timelapse.py.zip)

le-pet, 2023-10-02:
the last timelapse.py working good for me, just the quality is bad, and a error MIME in video preview on mainsail 

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/36859339/921bd3f4-5cab-4568-b189-a1cbed2257e1

BrianP525, 2023-10-02:
I found a few things with this one.
The video now render's correctly, and colors are correct, no artifacts.
I am able to play it back via fluidd interface, and locally when I download.
As @le-pet mentioned, the quality has significantly decreased. (Photos are still the same when exporting, just the render quality has dropped)

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/146669611/90c08a8b-abb3-41f4-aa9e-e55232a985d6

[image]

This file had the park printhead option toggled on by default, I was able to correct by disabling that option in the fluidd settings.

I also noticed that the print head is now randomly stopping every few layers for about 2-3 seconds, causing blobs on the surface. I put the original timelapse.py back to test, and it is still stopping.
I re-sliced the file without the timelapse gcode in the machine before layer change box, and still stopping.
Uninstalled fluidd, moonraker and nginx and the moonraker timelapse and re-installed all. Still stopping.
Not sure why it started doing that now, I dont think it would have been from the timelapse file, since putting the original back, and uninstalling/re-installing fluidd/moonraker did not work.
I did notice that in the system tab, it looks like Klipper load and MCU RPI load are alot higher than normal.
 
[image]

I am currently factory resetting and will test again once done.

Guilouz, 2023-10-02:
The only difference in two files, is the first use h264 encoding and the mast mjpg.

BrianP525, 2023-10-02:
I think my pausing issue is something else. I have downgraded firmware at this point and it is still doing it. I am leaning towards the file itself at this point.
As far as the timelapse, it works much better, but is there any way to increase the render quality? I took a look through that file, but didn't see a way to adjust quality.

Guilouz, 2023-10-02:
You can change render settings:

[image]

BrianP525, 2023-10-02:
I tried with setting Constant Rate Factor to 0 (which is supposed to be max quality) and 51, and dont see much difference. The render set to 0 is 6 bytes larger.
I'm not sure if you can see the video clips uploaded since they wont play here, but I grabbed a screenshot of what I am seeing. Both have large block-texture visible.
Constant Rate Factor set to 51:
[image]
Constant Rate Factor set to 0:
[image]

Guilouz, 2023-10-10:
I have improved timelapse render quality by improving bitrate.

closed 2023-10-10 by Guilouz

le-pet, 2023-10-11:
i try in the journey, thanks! :)