Guilouz/Creality-Helper-Script-Wiki #41: The Fluidd and Mainsail options no Longer work.

IriGitt, 2023-10-01:
After Installing Moonraker and Nginx, The script options to install Mainsail or Fluidd seem to crash/fail after entering y to confirm. Latest version of your script, Tried rebooting the machine. Not sure the issue of if its on me?

Guilouz, 2023-10-01:
Working fine for me, what is the error ?

IriGitt, 2023-10-02:
Hey mate It doesnt get an error. However I do know a little linux so I just went through your script and ran it manually and I was able to get Mainsail installed. This might be isolated to my machine. Feel free to close the issue I daresay its a 1 off on my end.

ak47h5n2, 2023-10-02:
Hello I got a error when I instailling Mainsail and Fluidd:

Installing Mainsail...
Connecting to github.com (20.205.243.166:443)
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
 Download failed. Exit code: 1

IriGitt, 2023-10-02:
For that error I recomend downloading a fresh copy of the script using

 cd && rm -rf /root/installer.sh && wget --no-check-certificate https://raw.githubusercontent.com/Guilouz/Creality-K1-and-K1-Max/main/Scripts/installer.sh

closed 2023-10-02 by Guilouz