Guilouz/Creality-Helper-Script-Wiki #48: Mainsail install error

setupisloading, 2023-10-04:
Getting this error while trying to install mainsail am in doing somthing wrong?

Installing Mainsail...
Connecting to github.com (140.82.114.4:443)
Connecting to github.com (140.82.114.4:443)
Connecting to objects.githubusercontent.com (185.199.108.133:443)
wget: can't open 'mainsail.zip': File exists
 Download failed. Exit code: 1

setupisloading, 2023-10-04:
Sorry i got it didnt see the option for resolved issues

Guilouz, 2023-10-04:
That's mean you already have a mainsail.zip file in /usr/data/

Download latest script i just updated with little fix for that.

closed 2023-10-04 by Guilouz

setupisloading, 2023-10-05:
Installing Moonraker and Nginx...
Downloading Moonraker and Nginx file...
Connecting to raw.githubusercontent.com (185.199.108.133:443)
saving to 'moonraker.tar'
moonraker.tar         99% |*******************************************************************************************************************************************************************************************  | 48.2M  0:00:00 ETAwget: short read, have only 13096
wget: connection closed prematurely
 Download failed. Exit code: 1

setupisloading, 2023-10-05:
Now i get this installing moonraker