Guilouz/Creality-Helper-Script-Wiki #51: Input shaping errors

seniordngdng, 2023-10-04:
Installed rooted firmware, moonraker, and fluidd following the guide.  did all the steps on the "fix issue with input shaper" page.  whenever i try to run Input_Shaper macro it goes through the whole process but at the very end I get these errors and have to restart the printer.  I did a factory reset and reloaded everything, but it still happens.

// The SAVE_CONFIG command will update the printer config file
// with these parameters and restart the printer.
12:34:17 // Klipper state: Shutdown
12:34:17 !! {"code":"key60", "msg":"Internal error on command:M84", "values": ["M84"]}
12:34:17 !! {"code":"key60", "msg":"Internal error on command:INPUT_SHAPER", "values": ["INPUT_SHAPER"]}

I looked in printer.cfg and it has the new values for recommended shaper type and freq, and they're different for x and y, like the example in the guide.

Remleyfpv, 2023-10-05:
Have the exact same thing: 
[image]

Guilouz, 2023-10-05:
I can’t reproduce this issue. Try to remove M84 command to see.

Remleyfpv, 2023-10-05:
[image]
 I removed M84 and it still doesnt work, do you want any files from me to check anything?

seniordngdng, 2023-10-05:
I think I figured it out... When I went through the guide and installed everything, there are parts that say to open printer.cfg and insert certain lines "at the end", which is what I did, but it actually needs to be inserted before the grey SAVE_CONFIG section. 
Everything seems to be working fine now.

Remleyfpv, 2023-10-05:
So how do I fix this now? update via the screen?

seniordngdng, 2023-10-05:
If you did what I described, you can either do a factory reset and install everything again or you can fix printer.cfg directly.

Remleyfpv, 2023-10-05:
[image]
Is it wierd that in my printer.cfg there  are a lot of "save config"s?

seniordngdng, 2023-10-05:
That's how mine was, so I used one of the original printer.cfg's that was saved as a backup (printer-20231004-xxxxx.cfg) from before I modified anything and basically started over.   Added the new code in the proper order, renamed that file printer.cfg and deleted the current printer.cfg, save and restart.  You'll have to redo you bed level and input shaper calibrations after that but doing that fixed the errors for me.

Remleyfpv, 2023-10-05:
Yess finaly worked after completely reinstalling all the firmware thanks guys!

closed 2023-10-05 by Guilouz