Guilouz/Creality-Helper-Script-Wiki #52: TLS Error installing mainsail or fluidd

Hook3D, 2023-10-05:
i keep getting this when using the helper script to install mainsail or fluidd

 Please enter your choice and validate with Enter: 3

 Are you sure you want to install Mainsail ? (y/n): y

Installing Mainsail...
Connecting to github.com (20.248.137.48:443)
wget: TLS error from peer (alert code 80): 80
wget: error getting response: Connection reset by peer
 Download failed. Exit code: 1

Hook3D, 2023-10-05:
fixed

found and old issue about the same thing

installed entware

rebooted

ran 

opkg install wget-ssl

then everything worked as normal

closed 2023-10-05 by Guilouz

QuinnDamerell, 2023-11-12:
I just hit this error on my factory fresh K1 max. I installed Moonraker with no issue using the script, but fluidd wouldn't install due to this error. I searched around and found this thread, did what @Hook3D did above, and it's all working now!

I see the script checks if moonraker is installed before Fluidd is installed. Maybe it should do the same with entware and the wget-ssl depdencies?

Guilouz, 2023-11-12:
wget-ssl is not a necessity, some have this error, others don't. And this is specified on my wiki.

QuinnDamerell, 2023-11-12:
@Guilouz interesting, I wonder why. That's good to know. Do you think it would be possible to check the return code of wget and print a link to the help doc if it fails?

Guilouz, 2023-11-12:
It's a bit complicated, because it's really random, I haven't found the reason why some people have this error (maybe the country in which it is located). 
People are expected to read the wiki before use.
 I will see if it is possible to search for this and invite people to go to the wiki to correct the error

QuinnDamerell, 2023-11-13:
Yeah, that's a good point. I was following a different guide on a different website that didn't point to the wiki. The other odd thing was when I searched the error string "wget: TLS error from peer (alert code 80): 80" I didn't really find anything in Google about it, it was strange. I had the idea to search the issues on the project, which is where I found it.

Anyways, that all makes sense, thanks for discussion it with me! If I can help you debug it in any way, please let me know!

(p.s. I was wondering if it might be location-dependent as well. I'm sure Github has many POPs and maybe some of them have certs that are not compatible with wget for some reason. Maybe some POPs use a signing root that wasn't trusted back when that version of wget was built? I'm in the Seattle area, if you jumped on a VPN and moved to Seattle, I wonder if it would repo.)

QuinnDamerell, 2023-11-13:
I took at the cert chain I get from Seattle when I go to:

https://github.com/fluidd-core/fluidd/releases/download/v1.26.2

The root cert has a "valid not before" of 2006, but the secondary cert has a "valid not before" date of 2021. I wonder if that's the problem? I noticed the version of the base OS on the K1 reports to be built in 2020

[image]