Guilouz/Creality-Helper-Script-Wiki #53: KAMP doesnt do a smaller bed mesh but just a regular one

Remleyfpv, 2023-10-05:
After following this guide and installing KAMP the K1 does a  full bed mesh every print. (I use orca slicer so I believe it should work no slicer changes right? as  I also see the"exclude object" button in mainsail)

Guilouz, 2023-10-05:
You have missing something.

Do you have enabled enable_object_processing in moonraker.conf?

joschiv1977, 2023-10-05:
Label objects must be on in your slicer

[image]

Remleyfpv, 2023-10-05:
And 

> You have missing something.
> 
> Do you have enabled enable_object_processing in moonraker.conf?
[image]

joschiv1977, 2023-10-05:
> How do you do that in orcaslicer?

Search is the key 

[image]

[image]

[image]

Remleyfpv, 2023-10-05:
Fixed! had to completely reload all firmware!

closed 2023-10-05 by Remleyfpv