Guilouz/Creality-Helper-Script-Wiki #55: KAMP and max_accel value

joschiv1977, 2023-10-06:
After enable KAMP, smart park and line purge i noticed my max_accel values are stuck at 500 (it think its set like at bed mesh and other stuff). I set it at 10000 in my printer.cfg 

After installing firmware new, and installed everything from ground up, problem still exists. Disable KAMP will leave my max_accel untouched at 10000

Disabled smart park and line purge, just KAMP. Same problem. Have to disable KAMP to get my max_accel values after bed level. 

To get working for me now i added SET_VELOCITY_LIMIT ACCEL=10000 to my start_print macro

Guilouz, 2023-10-06:
Very strange because KAMP doesn't touch to max_velocity param.

joschiv1977, 2023-10-06:
After a litte bit of research, i found the problem in the /usr/share/klipper/klippy/extras/custom_macro.py. 

        if self.pheaters.can_break_flag == 3:
            self.pheaters.can_break_flag = 0
            self.gcode.respond_info("can_break_flag is 3")
            self.gcode.run_script_from_command('M220 S100')
            self.gcode.run_script_from_command('M221 S100')
            self.gcode.run_script_from_command('LINE_PURGE')

In the creality file this is after the purge line ?! and what is missing in your file: 
            self.gcode.run_script_from_command('M82')
            self.gcode.run_script_from_command('G92 E0')
            # self.gcode.run_script_from_command('G1 Z2.0 F600')
            self.gcode.run_script_from_command('G1 F12000')
            self.gcode.run_script_from_command('G21')

Guilouz, 2023-10-06:
It's ok ?

joschiv1977, 2023-10-06:
no. i was thinking the custom_marco.py was the problem. But with KAMP only (without smart park and Line purge) the problem still exists.

joschiv1977, 2023-10-06:
i can see in the klipper log file that the value is changing during mesh and all those stuff to 500 but after line purge and heating the nozzle to the print temp the original value will not be loaded. So there must be a problem in some other script or macro.

Guilouz, 2023-10-07:
As said on Discord, I have done many test and I can't reproduce issue, after purge, max accel id done by value configured in my slicer for first layer without issue.

When calibration:

[image]

When starting first layer:

[image]

closed 2023-10-07 by Guilouz