Guilouz/Creality-Helper-Script-Wiki #56: KAMP issue

bhast2, 2023-10-07:
When I try to use Adapted mesh I get this error after it gets done creating the mesh.
```
08:12:49 AM 
!! {"code":"key60", "msg":"Internal error on command:_BED_MESH_CALIBRATE", "values": ["_BED_MESH_CALIBRATE"]}
08:12:49 AM 
!! {"code":"key60", "msg":"Internal error on command:BED_MESH_CALIBRATE", "values": ["BED_MESH_CALIBRATE"]}
08:12:49 AM 
!! {"code":"key60", "msg":"Internal error on command:START_PRINT", "values": ["START_PRINT"]}
```

I have followed the steps in the wiki and still get this error.

Guilouz, 2023-10-07:
You must have missed something.

Uninstall KAMP and reinstall it, I simplified the installation in my latest version of the script that I just released. And see updated [Use KAMP](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Use-KAMP) section.

kwotattoos, 2023-10-08:
[image]

i uninstalled and reinstalled kamp and this is the error im getting

Guilouz, 2023-10-08:
Remove kamp from update manager in moonraker.conf, it must not be updated with the fix.

closed 2023-10-08 by Guilouz