Guilouz/Creality-Helper-Script-Wiki #58: Stuck on self test screen

setupisloading, 2023-10-08:
I messed up trying to start with a fresh root. tried to go back to stock firmware to redo the root now my screen is stuck on the first boot screen where you pick the language and self test the printer i go through the menus and it fails the self test wont let me get to the main screen. Weird thing is mainsail works fine and i could use the with no problems with mainsail but if possible i would like my screen back can anybody help?

setupisloading, 2023-10-08:
Also i can SSH but the default root creality_2023 is not working.

Guilouz, 2023-10-08:
It’s a printer/firmware issue, not related to wiki.

If ssh password don’t work that’s mean you have not enabled root access

Please post only issue related to the wiki.

closed 2023-10-08 by Guilouz