Guilouz/Creality-Helper-Script-Wiki #60: Unkno

nephilim667, 2023-10-09:


nephilim667, 2023-10-09:
Hello,

I have followed all the wiki to root my K1
All works well except for KAMP. 
I'm using PrusaSlicer and followed the tutorial, but when print is launched there's some unrecognized macro.
I have made some research on internet and it seem these macro missing somewhere.

Here is the missed macro :
## exclude object macros
[gcode_macro DEFINE_OBJECT]
gcode: EXCLUDE_OBJECT_DEFINE {rawparams}
[gcode_macro START_CURRENT_OBJECT]
gcode: EXCLUDE_OBJECT_START NAME={params.NAME}
[gcode_macro END_CURRENT_OBJECT]
gcode: EXCLUDE_OBJECT_END {% if params.NAME %}NAME={params.NAME}{% endif %}
[gcode_macro LIST_OBJECTS]
gcode: EXCLUDE_OBJECT_DEFINE
[gcode_macro LIST_EXCLUDED_OBJECTS]
gcode: EXCLUDE_OBJECT
[gcode_macro REMOVE_ALL_EXCLUDED]
gcode: EXCLUDE_OBJECT RESET=1

i have made a KAMP_macro.cfg file and include it in printer.cfg. After what all seem to be working as it should.

Another issue i have discovered tonight,
When i have calibrated the printer, i have definined a 12x12 mesh with bilinear algorithm and tension equal to 0.1.
when launching a print from fluidd, KAMP adjust mesh as it should. (for example, it define a 5x5 mesh instead of using the 12x12 in printer.cfg file) but when launching print directly from K1 screen, printer does a full bed mesh (12x12)

Guilouz, 2023-10-09:
I think you use Prusa Slicer that it use DEFINE_OBJECT instead of EXCLUDE_OBJECT used by Klipper. That explains missing macros.

closed 2023-10-10 by Guilouz