Guilouz/Creality-Helper-Script-Wiki #65: Timelapse weird colors

Kirbo, 2023-10-14:
I know you haven't wrote the Timelapse code or anything, but I was just wondering as you perhaps might have K1 Max as well (or K1 with camera), that have you experienced the similar issue and what should be done to fix it?
The timelapses themselves are as they should, imo, but the colors seem to be weird.
The camera stream and the snapshots look nothing like the timelapse that is being automatically rendered after the print.

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/1077974/0866034d-b3b5-4f85-87bb-d807ec868cb1

The print itself should be RED, but on the timelapse it looks like its white/grey.

I haven't touched the Timelapse nor the camera settings at all, I just followed the instructions here https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Use-Timelapse and the video attached here is what I got

Guilouz, 2023-10-14:
Just remove timelapse and reinstall it with lat st script version, i have fixed it.

closed 2023-10-14 by Guilouz

Kirbo, 2023-10-14:
@Guilouz Updated the `installer.sh`, removed the `Moonraker Timelapse` via the installer script and re-installed it. Issue persists, please advice.

[image]
[image]
[image]

https://github.com/Guilouz/Creality-K1-and-K1-Max/assets/1077974/016e13e9-536b-4015-bb80-dc1ee9c15862

`timelapse.cfg`
```yaml
# Timelapse klipper macro definition
#
# Copyright (C) 2021 Christoph Frei <fryakatkop@gmail.com>
# Copyright (C) 2021 Alex Zellner <alexander.zellner@googlemail.com>
#
# This file may be distributed under the terms of the GNU GPLv3 license
#
# Macro version 1.14
#

##### DO NOT CHANGE ANY MACRO!!! #####

##########################################################################
#                                                                        #
# GET_TIMELAPSE_SETUP: Print the Timelapse setup to console              #
#                                                                        #
##########################################################################

[gcode_macro GET_TIMELAPSE_SETUP]
description: Print the Timelapse setup
gcode: 
  {% set tl = printer['gcode_macro TIMELAPSE_TAKE_FRAME'] %}
  {% set output_txt = ["Timelapse Setup:"] %}
  {% set _dummy = output_txt.append("enable: %s" % tl.enable) %}
  {% set _dummy = output_txt.append("park: %s" % tl.park.enable) %}
  {% if tl.park.enable %}
    {% set _dummy = output_txt.append("park position: %s time: %s s" % (tl.park.pos, tl.park.time)) %}
    {% set _dummy = output_txt.append("park cord x:%s y:%s dz:%s" % (tl.park.coord.x, tl.park.coord.y, tl.park.coord.dz)) %}
    {% set _dummy = output_txt.append("travel speed: %s mm/s" % tl.speed.travel) %}
  {% endif %}
  {% set _dummy = output_txt.append("fw_retract: %s" % tl.extruder.fw_retract) %}
  {% if not tl.extruder.fw_retract %}
    {% set _dummy = output_txt.append("retract: %s mm speed: %s mm/s" % (tl.extruder.retract, tl.speed.retract)) %}
    {% set _dummy = output_txt.append("extrude: %s mm speed: %s mm/s" % (tl.extruder.extrude, tl.speed.extrude)) %}
  {% endif %}
  {% set _dummy = output_txt.append("verbose: %s" % tl.verbose) %}
  {action_respond_info(output_txt|join("\n"))}

################################################################################################
#                                                                                              #
# Use _SET_TIMELAPSE_SETUP [ENABLE=value] [VERBOSE=value] [PARK_ENABLE=value] [PARK_POS=value] #
#                          [PARK_TIME=value] [CUSTOM_POS_X=value] [CUSTOM_POS_Y=value]         #
#                          [CUSTOM_POS_DZ=value][TRAVEL_SPEED=value] [RETRACT_SPEED=value]     #
#                          [EXTRUDE_SPEED=value] [EXTRUDE_DISTANCE=value]                      #
#                          [RETRACT_DISTANCE=value] [FW_RETRACT=value]                         #
#                                                                                              #
################################################################################################

[gcode_macro _SET_TIMELAPSE_SETUP]
description: Set user parameters for timelapse
gcode:
  {% set tl = printer['gcode_macro TIMELAPSE_TAKE_FRAME'] %}
  ##### get min and max bed size #####
  {% set min = printer.toolhead.axis_minimum %}
  {% set max = printer.toolhead.axis_maximum %}
  {% set round_bed = True if printer.configfile.settings.printer.kinematics is in ['delta','polar','rotary_delta','winch'] 
                else False %}
  {% set park = {'min'   : {'x': (min.x / 1.42)|round(3) if round_bed else min.x|round(3),
                            'y': (min.y / 1.42)|round(3) if round_bed else min.y|round(3)},
                 'max'   : {'x': (max.x / 1.42)|round(3) if round_bed else max.x|round(3),
[... 342 lines cut ...]
##########################################################################
#                                                                        #
# TEST_STREAM_DELAY: Helper macro to find stream and park delay          #
#                                                                        #
##########################################################################

[gcode_macro TEST_STREAM_DELAY]
description: Helper macro to find stream and park delay
gcode:
  {% set min = printer.toolhead.axis_minimum %}
  {% set max = printer.toolhead.axis_maximum %}
  {% set act = printer.toolhead.position %}
  {% set tl = printer['gcode_macro TIMELAPSE_TAKE_FRAME'] %}
  {% if act.z > 5.0 %}
    G0 X{min.x + 5.0} F{tl.speed.travel|int * 60}
    G0 X{(max.x-min.x)/2}
    G4 P{tl.park.time|float * 1000}
    _TIMELAPSE_NEW_FRAME HYPERLAPSE=FALSE
    G0 X{max.x - 5.0}
  {% else %}
    {action_raise_error("Toolhead z %.3f to low. Please place head above z = 5.0" % act.z)}
  {% endif %}
```

Any other information I could provide?

Guilouz, 2023-10-14:
Can you send your timelapse file

Kirbo, 2023-10-14:
> Can you send your timelapse file

Which file you mean? The video file or the `timelapse.cfg`? Both can be found from my previous message.

Guilouz, 2023-10-14:
Timelapse video file. It’s mov file, it’s not original file

Kirbo, 2023-10-15:
> Timelapse video file. It’s mov file, it’s not original file

Yes, I had to scale it down to 720p, as GitHub allows max 10MB files. I’ll share one originaö later today, not at home atm.

Kirbo, 2023-10-15:
@Guilouz here you go, the original Timelapse file: https://www.dropbox.com/scl/fi/lxm2g205kkfhkbvhoonp6/timelapse_OrcaCube_PLA_59m27s_300mms_false_35_top_standard.gcode_2023-10-14_2216.mp4?rlkey=dnahlsgo01mqdu2640ghwyqtm&dl=0

Guilouz, 2023-10-15:
Your Timelapse use H264 encoding so you don't use my latest timelapse version it use mjpeg.

With h264 for unknown reason there is this kind of artefacts.

Kirbo, 2023-10-15:
> Your Timelapse use H264 encoding so you don't use my latest timelapse version it use mjpeg.
> 
> With h264 for unknown reason there is this kind of artefacts.

Yeah, I just noticed that I might have forgotten to restart in between. I just like 5 minutes ago removed and re-installed it again, restarted the whole machine, noticed that the "Timelapse" is back in Mainsail and now I'm printing. Hoping the best it'd work now!

Kirbo, 2023-10-15:
Now it worked, thanks!
https://www.dropbox.com/scl/fi/9w3j28ex65leftctj4sm5/timelapse_OrcaCube_PLA_1h1m_300mms_false_35_top_standard.gcode_2023-10-15_2206.mp4?rlkey=x6cr3qmroh921n064j4en7vn3&dl=0

Guilouz, 2023-10-15:
Perfect