Guilouz/Creality-Helper-Script-Wiki #66: Move out of range error after KAMP

The760, 2023-10-14:
Just want to say that I love what you've done here. Thanks for putting together the Wiki.

Unfortunately, I am having a move out of range error once KAMP completes. The Mesh Bed Levelng procedure completers and saves to [default], but it fails when it goes to park the toolhead for purging.

[image]

I assume KAMP is trying to park the toolhead out of range?

Guilouz, 2023-10-14:
Do you use latest version of KAMP i have updated?

The760, 2023-10-14:
> Do you use latest version of KAMP i have updated?

I only just installed your stuff for the first time half an hour ago, so I hope so! Lol

This is on the standard K1 if that helps. Not the Max.

The760, 2023-10-15:
I've tried a bunch of different stuff but have not found a fix. This print taakes up most of the build plate, and it seems the smart park is trying to go out of bounds.

Kirbo, 2023-10-16:
I found out that some prints (especially Pressure Advance and First Layer tests) don't work with KAMP.
Other than these, KAMP has worked perfectly for me since the `x <> y` fix.
For the tests, I just turn off the KAMP and I'm able to print. Would be awesome if even these worked, but not that big of a deal for me.

joschiv1977, 2023-10-16:
Same here:
{"code":"key243","msg":"Move out of range: 54.499 -1.000 10.100 [0.000]", "values":[54.499, -1.000, 10.100, 0.000]}
14:42
{"code":"key243","msg":"Move out of range: 54.499 -1.000 10.100 [0.000]", "values":[54.499, -1.000, 10.100, 0.000]}
14:42
{"code":"key243","msg":"Move out of range: 54.499 -1.000 10.100 [0.000]", "values":[54.499, -1.000, 10.100, 0.000]}

Got the error with this layout:

[image]

After ich change it to this layout KAMP works without any problem. Maybe smart park is the problem. 

[image]

joschiv1977, 2023-10-16:
variable_purge_margin: 10                   # Distance the purge will be in front of the print area, default is 10.

This is the problem. So it will be added and my distance is 9 and it adds 10. So -1 is out of range. 
Maybe i play with the bed mesh values.

The760, 2023-10-16:
> variable_purge_margin: 10 # Distance the purge will be in front of the print area, default is 10.
> 
> This is the problem. So it will be pulled of and my distance is 9 and it pulls off 10. So -1 is out of range. Maybe i play with the bed mesh values and or bed size values ?!

Let me know if you figure it out. My print also covers nearly the entire Y axis like yours. The only way I've been able to start a print is disabling both line purge and smart park. This results in no purge though, so there's a high chance the nozzle oozes on the plate before the print begins.

ACKET2009, 2023-10-19:
The problem of the moving out of range I notice, when trying to use timelapse. The head kick to the far left corner after the first layer printed and the layers shifted on the second layer. The timelapse was installed from this script... And I remove the timelapse from the config

Guilouz, 2023-10-23:
Can you try this files by replacing by existing one :
 
[New KAMP.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13077998/New.KAMP.zip)

The760, 2023-10-25:
[image]

Still getting the same error with Smart Park enabled.

Ramsin, 2023-10-28:
I used both Orca and Creality Print, and I only have this issue using Creality Print. Another thing that should be added to the tutorial is, when in Creality Print, under the Parameter Config -> Experimental, the "Exclude Objects" should be checked. This was the only thing that got it to actually do the localized meshing. I know that you have it in the wiki, but it's suggested for Prusa Slicer, but applies to Creality print as well.

Ramsin, 2023-10-29:
> I used both Orca and Creality Print, and I only have this issue using Creality Print. Another thing that should be added to the tutorial is, when in Creality Print, under the Parameter Config -> [...]

Update: As stated above, I *was* running into this issue using Creality Print (macOS) after installing KAMP. However, it seems to have been remedied by, in addition to the aforementioned "Exclude Objects" needing to be checked, you also have to check/enable "Relative Extrusion" in Parameter Config -> Special Modes (with the "Advanced" toggled turned on).

Nastras, 2023-10-31:
https://www.reddit.com/r/crealityk1/comments/17ka5qg/fix_for_kamp_out_of_range_error/

Guilouz, 2023-10-31:
Someone can test this files and tell me if all is ok ?

1. Uninstall previous version with my script.
2. Download [KAMP.zip](https://github.com/Guilouz/Creality-K1-and-K1-Max/files/13221483/KAMP.zip)
3. Copy KAMP folder in your Klipper files folder.
4. Move KAMP_Settings.cfg file in root of your Klipper files folder (where printer.cfg is)
5. Enable enable_object_processing: True in moonraker.conf
6. Add [include KAMP_Settings.cfg] in printer.cfg
7. Replace START_PRINT macro by this one:

  ```
  [gcode_macro START_PRINT]
  variable_prepare: 0
  gcode:
    WAIT_TEMP_END
    CLEAR_PAUSE
    {% set g28_extruder_temp = printer.custom_macro.g28_ext_temp %}
    {% set bed_temp = printer.custom_macro.default_bed_temp %}
    {% set extruder_temp = printer.custom_macro.default_extruder_temp %}
    {% if 'BED_TEMP' in params|upper and (params.BED_TEMP|float) %}
    {% set bed_temp = params.BED_TEMP %}
    {% endif %}
    {% if 'EXTRUDER_TEMP' in params|upper and (params.EXTRUDER_TEMP|float) %}
    {% set extruder_temp = params.EXTRUDER_TEMP %}
    {% endif %}
  
    {% if printer['gcode_macro START_PRINT'].prepare|int == 0 %}
      {action_respond_info("not prepare.\n")}
      PRINT_PREPARE_CLEAR
      CX_ROUGH_G28 EXTRUDER_TEMP={extruder_temp} BED_TEMP={bed_temp}
      CX_NOZZLE_CLEAR
      ACCURATE_G28
      _KAMP_CALIBRATION
    {% else %}
      PRINT_PREPARE_CLEAR
    {% endif %}
    _KAMP_PURGE
  ```

Guilouz, 2023-11-01:
Issue has been fixed with latest script v3.2.

closed 2023-11-01 by Guilouz