Guilouz/Creality-Helper-Script-Wiki #68: MobileRaker

elpigeondindo, 2023-10-16:
salut, deja super taf pour tous ces scripts reunis au meme endroit.

J'ai voulu installer mobileraker companion, tout se passe bien, mais quand je vais sur mon smartphone il arrive pas a se connecter a l'imprimante ou au socket. Il faut faire un truc en plus pour le faire fonctionner ?

Quelqu'un a deja reussi?
-------------------------------------------
hi, already great work for all these scripts gathered in the same place.

I wanted to install mobileraker companion, everything goes well, but when I go to my smartphone it cannot connect to the printer or the socket. Do I have to do something extra to make it work?

Has anyone already succeeded?

joschiv1977, 2023-10-16:
you can open the url http://192.168.1.30:4409/ in your webbrowser on your phone ?

elpigeondindo, 2023-10-16:
yes of course

joschiv1977, 2023-10-16:
try https://192.168.1.30:4409 and make a check mark at trust self signing certificates in the mobileraker app

elpigeondindo, 2023-10-16:
i can't acces to mainsail on HTTPS

joschiv1977, 2023-10-16:
do it in the app

elpigeondindo, 2023-10-16:
i try and i have this

connection refused (os error: connection refused, errno = 61), address = 192.168.1.30, port 49654

elpigeondindo, 2023-10-16:
[https://i.ibb.co/S72W7Gg/IMG-2412.png](url)

a screenshot

Guilouz, 2023-10-16:
Yu as oublié les ports dans tes adresses.

elpigeondindo, 2023-10-16:
Oh put** quel nul ! comment j'ai pu oublié ca lol. Merci guillouz

closed 2023-10-17 by elpigeondindo