Guilouz/Creality-Helper-Script-Wiki #70: Random pause

lvlacleod, 2023-10-17:
Not sure if it's anything to do with your scripts to be honest but the last 2 days my prints have paused after the purge line. It's only happened 4 or 5 times where I have ran maybe 20 prints.
The last thing I changed was adding firmware retraction in gcode_macro.cfg

lvlacleod, 2023-10-19:
Update. Removed Automatic saving but it's still pausing and I think it has something to do with the file size of the gcode. I am currently going through a lithophane phase and most of the files are over 30MB. Everything else I've printed before I started on lithophanes never paused after the purge line.
Do you have any idea what would be causing this? Help would be greatly appreciated. Thanks

Guilouz, 2023-10-19:
Not related to the Wiki.

Please only post issue related to the wiki here.

closed 2023-10-19 by Guilouz