Guilouz/Creality-Helper-Script-Wiki #73: input_shaper

ludo453, 2023-10-20:
Bonjour Cyril, 

Avec le dernier firmware j'ai les messages ci-dessous dans le input_shaper

!! {"code":"key60", "msg":"Internal error on command:M84", "values": ["M84"]}
15:04:03 
!! {"code":"key60", "msg":"Internal error on command:INPUT_SHAPER", "values": ["INPUT_SHAPER"]}

Nazgile94, 2023-10-20:
INPUT_SHAPER
is the custom macro 

---
i translating 👍

Hi Cyril,

With the latest firmware I have the following messages in the input_shaper

!! { "code": "key60", "msg": "Internal error on command:M84", "values": ["M84"]}
15:04:03
!! { "code": "key60", "msg": "Internal error on command:INPUT_SHAPER", "values": ["INPUT_SHAPER"]}

I just saw that now I have INPUTSHAPER and INPUT_SHAPER now which one to take?

Guilouz, 2023-10-20:
You must use INPUT_SHAPER macro to run test manually.

ludo453, 2023-10-20:
Ok it's him I take and he gives me the mistakes

closed 2023-10-20 by ludo453