Guilouz/Creality-Helper-Script-Wiki #75: orca pla profile not working

bodydropped72, 2023-10-21:
ive been trying to get a print to start using your pla profile and it gives me this error EVERY time   

13:18:08 
// {"code":"key111", "msg": "Extrude below minimum temp
// See the 'min_extrude_temp' config option for details", "values": []}
13:18:08 
!! {"code":"key111", "msg": "Extrude below minimum temp

also why does the extruder heat up then it homes to center of build plate then it cools down a significant amount then heats up again. this is where i believe the error is happening. ive followed your instruction to the T. The ABS profile works but it also does that thing where the extruder will heat up, then cool down then heat up again

Guilouz, 2023-10-22:
This is how Creality Os works, temp is reduced during calibration. I have no issue with PLA.

closed 2023-10-22 by Guilouz

bodydropped72, 2023-10-22:
im using all of your profiles. every single time once it goes through
operating sequence and the print gets ready to start it stops on extruder
temp to low. i have not changed anything. Literally copy/pasted all of your
stuff. Ive printed with abs just fine btw

aaronsmith1234, 2023-11-15:
As a note, I've had this same issue; used the profiles as provided, had exact same issues.... @Guilouz happy to help debug

aaronsmith1234, 2023-11-15:
As a note, I added 
`min_extrude_temp: 170`
to the [extruder] section in printer.cfg, and had no change. The message still says:

`{"code":"key111", "msg": "Extrude below minimum temp
See the 'min_extrude_temp' config option for details", "values": []}`

which makes me think its not picking the value up correctly...

aaronsmith1234, 2023-11-15:
@bodydropped72 were you able to resolve?

bodydropped72, 2023-11-15:
If I remember correctly I went back to original root. Then I did something and went back but using a different profile in orca. Haven’t used machine in over a week due to hotend board failure. I’d have to check when I get it back up and runningRobby Mendel

aaronsmith1234, 2023-11-15:
And as a note I am using OrcaSlicer v 1.8.0-rc2, and am on firmware version 1.3.2.1

aaronsmith1234, 2023-11-15:
I can also confirm that if I use a different slicer\profile (namely PrusaSlicer), things work fine, so its either the slicer or the profile... @Guilouz what version of OrcaSlicer are you using?