Guilouz/Creality-Helper-Script-Wiki #81: Proposal for Knomi macro integration

nephilim667, 2023-10-25:
Hello,

I have integrated a Knomi from BigTreeTech with the D3vil design head mount.

Actually in Knomi documentation they tell to add relevant macros for homing and bed leveling.
Unfortunatly those macro aren't working on K1 !

After some research i have finally make it work.

for homing, if you put macro as it it will throw an error 'G28 is called recursively'.
I have found that come from macro defined in sensorless.cfg

So to make it work, first add the macro at end of file
[gcode_macro G28]
rename_existing: G0028
variable_homing:False
gcode:
  RESPOND TYPE=command MSG='Custom G28 called'
  SET_GCODE_VARIABLE MACRO=G28 VARIABLE=homing VALUE=True
  G0028 {rawparams}
  SET_GCODE_VARIABLE MACRO=G28 VARIABLE=homing VALUE=False

atfer what find _HOME_X, _HOME_Y and _HOME_Z macro and replace G28 in it by G0028 like this
[gcode_macro _HOME_X]
gcode:
  _IF_MOVE_XY

  {% if printer['gcode_macro xyz_ready'].x_ready|int == 1 %}
    {% if (printer.configfile.settings['stepper_x'].position_max - printer.toolhead.position.x)|round < 10 %}
      {% set x_park = (10 - (printer.configfile.settings['stepper_x'].position_max - printer.toolhead.position.x))|round %}
      {% if x_park > 0 %}
        G91
        G1 x-{x_park} F3600
        G90
        G4 P1000
      {% endif %}
    {% endif %}
  {% endif %}
  
  # SET_TMC_FIELD FIELD=SGTHRS STEPPER=stepper_y VALUE=70
  # SET_TMC_FIELD FIELD=SGTHRS STEPPER=stepper_x VALUE=70
  # Home
  RESPOND TYPE=command MSG='Call G28 in _HOME_X macro'
  G0028 X #<<====== HERE
  SET_GCODE_VARIABLE MACRO=xyz_ready VARIABLE=x_ready VALUE=1
  # Move away
  G91
  G1 x-10 F3600
  G90
  # Wait just a second (give StallGuard registers time to clear)
  G4 P2000

For bed leveling, as i'm using Kamp, in Adaptative_Meshing.cfg, BED_MESH_CALIBRATE Macro is already defined
So just add  "variable_probing:False" after rename existing and SET_GCODE_VARIABLE before and after calling _BED_MESH_CALIBRATE

Here is the modified macro :
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: _BED_MESH_CALIBRATE
variable_probing:False
gcode:

    {% set all_points = printer.exclude_object.objects | map(attribute='polygon') | sum(start=[]) %}                                # Gather all object points
    {% set bed_mesh_min = printer.configfile.settings.bed_mesh.mesh_min %}                                                          # Get bed mesh min from printer.cfg
    {% set bed_mesh_max = printer.configfile.settings.bed_mesh.mesh_max %}                                                          # Get bed mesh max from printer.cfg
    {% set probe_count = printer.configfile.settings.bed_mesh.probe_count %}                                                        # Get probe count from printer.cfg
    {% set kamp_settings = printer["gcode_macro _KAMP_Settings"] %}                                                                 # Pull variables from _KAMP_Settings
    {% set verbose_enable = kamp_settings.verbose_enable | abs %}                                                                   # Pull verbose setting from _KAMP_Settings
    {% set probe_dock_enable = kamp_settings.probe_dock_enable | abs %}                                                             # Pull probe dockable probe settings from _KAMP_Settings
    {% set attach_macro = kamp_settings.attach_macro | string %}                                                                    # Pull attach probe command from _KAMP_Settings
    {% set detach_macro = kamp_settings.detach_macro | string %}                                                                    # Pull detach probe command from _KAMP_Settings
[... 87 lines cut ...]
            )) }

            {action_respond_info("KAMP adjustments successful. Happy KAMPing!")}

        {% else %}

            {action_respond_info("No objects detected! Check your gcode and make sure that EXCLUDE_OBJECT_DEFINE is happening before BED_MESH_CALIBRATE is called. Defaulting to regular meshing.")}
            G4 P5000                                                                                                                # Wait 5 seconds to make error more visible
        {% endif %}

    {% endif %}

    {% if probe_dock_enable == True %}
        {attach_macro}                                                                                                              # Attach/deploy a probe if the probe is stored somewhere outside of the print area
    {% endif %}

    SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=probing VALUE=True
    _BED_MESH_CALIBRATE mesh_min={adapted_x_min},{adapted_y_min} mesh_max={adapted_x_max},{adapted_y_max} ALGORITHM={algorithm} PROBE_COUNT={points_x},{points_y}
    SET_GCODE_VARIABLE MACRO=BED_MESH_CALIBRATE VARIABLE=probing VALUE=False

    {% if probe_dock_enable == True %}
        {detach_macro}                                                                                                              # Detach/stow a probe if the probe is stored somewhere outside of the print area
    {% endif %}                                                                                                                     # End of verbose

Hope you'll find this usefull

mindkill666, 2023-10-25:
Nice, i have a knomi, going to try this method

nephilim667, 2023-10-25:
i propose this way because i have try multiple time to flash my knomi, but unfortunatly platformio won't connect to it and so it's impossible to me to use the PR proposed on BTT Knomi github.

Guilouz, 2023-10-25:
> i propose this way because i have try multiple time to flash my knomi, but unfortunatly platformio won't connect to it and so it's impossible to me to use the PR proposed on BTT Knomi github.

You can find how to flash firmware on my other wiki here : https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad/wiki/Use-BigTreeTech-KNOMI

closed 2023-10-26 by Guilouz