Guilouz/Creality-Helper-Script-Wiki #82: Backup in script 3.1 doesn't work

Nastras, 2023-10-26:
I noticed that the backup function in the script only creates a file with 0 bytes and no content.

Has there been an error?

[image]

Guilouz, 2023-10-26:
There is no change in backup since many updates.
What is the result when backup in script ?

Nastras, 2023-10-26:
No free memory. I missed the message. The question is why is there no memory?

[image]
[image]

Guilouz, 2023-10-26:
It's you root partition is full. I don't know what exactly you did to get there but I think that to find something correct you will have to reset to factory

Nastras, 2023-10-27:
Thanks for the tip. I am surprised that the memory for root is used up because I have nothing there except a mini script. The only thing that is there additionally is data from the usb cam tool.

I will check it later when I am at home and report.

Nastras, 2023-10-27:
You were right. I have now deleted the USB Cam Tool which is stored in the root folder. After that I could make the backup normally because there is enough space again. I think the tool should not be stored in the root folder if the space is so small.

Would it actually be possible that you integrate the root folder into the backup for your own scripts?

closed 2023-10-27 by Nastras