Guilouz/Creality-Helper-Script-Wiki #84: I can't connect via SSH

jacquesglerner, 2023-10-27:
I updated to FW 1.3.2.1 and installed 'Helper'. That was not brilliant but let's say it was OK. My friend, who also uses the same machine, watched a new version of FW 1.3.2.8  and pushed it to the next firmware. After some errors, I managed to reboot and get 1.3.2.8 working. I've lost my Fluidd and, I think,  a little bit more. So, I re-seted it to factory defaults.
Now, I don't have any 'Klipper' working but, I managed to use my K1 printer in factory conditions.
My problem now is I don't get any, and I mean any SSH to the IP address of my K1, You name it: Putty Mobaxterm, Bitvise...
Sometimes I'm able to punch "root" and "creality_2023" (as I re-rooted the printer). But  I stop here. No matter any SSH program I use the printer does not accept SSH.  I'm stopped here. Any suggestions?

Guilouz, 2023-10-27:
you must re-validate root access from the screen To be able to access ssh again.

closed 2023-10-29 by Guilouz

jacquesglerner, 2023-10-30:
@Guilouz, I have already done the re-validation., even before you said so. Even with that, despite the printer being just beside another Ender3 v2 running under a Sonic Pad, using "Advanced IP Scanner", sometimes I get the K1 IP, others don't. 
Same thing with SSH, sometimes I'm able to connect but the MB seems with 'no connection' or "limited connection".
That's why I think there's something related to the way the motherboard connects to WiFi. Just to let you know I did it  2 times. The first was the general Net from myhome. The second one I created a sub-net 2.4G only and did the same re-validation. Actually both show the same issues.

shrimmy, 2024-01-12:
I have the same issue with "flaky" wifi reliability on the k1 after boot. When wifi does work it works flawlessly, but sometimes it does not start at all