Guilouz/Creality-Helper-Script-Wiki #90: Klipper Input Shaping Assistant is not only for Windows

nevack, 2023-10-31:
[This wiki section](https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Fix-issue-with-Input-Shaper#extra) states, that [Klipper Input Shaping Assistant](https://github.com/theycallmek/Klipper-Input-Shaping-Assistant/) is only for Windows, but I have successfully ran it on macOS.

<details>
<summary>Proof:</summary>

[image]
[image]

</details>

<details>
<summary>Requires some python knowledge, but pretty much easy to do</summary>

```
# Install Python with TkInter (used 3.10 just in case)
brew install python@3.10 python-tk@3.10

# Clone Repo and cd into it
git clone https://github.com/theycallmek/Klipper-Input-Shaping-Assistant.git kisa && cd kisa

# Create and activate venv
python3.10 -m venv venv
source venv/bin/activate

# Install Klipper-Input-Shaping-Assistant deps
pip install numpy matplotlib

# Run it!
python main.py
```

</details>

Guilouz, 2023-10-31:
Yes I have specified for Windows because there is a compiled exe file, more easier for people (Input-Shaper-Calibration-Assistant.exe) but off course python can be run on other OS.

nevack, 2023-11-01:
> Yes I have specified for Windows because there is a compiled exe file, more easier for people (Input-Shaper-Calibration-Assistant.exe) but off course python can be run on other OS.

Yep, I understand, but maybe it's worth to clarify that prebuilt binaries are only available on Windows, but it can be run from source on other OS.

nevack, 2023-11-01:
Maybe unrelated to the issue, but another good thing to add would be how to pull this calibration data from device.
`scp` didn't work for me, as sshd is not configured for it.
I have used something like this:
```sh
ssh root@creality 'cat /tmp/calibration_data_y_20231031_134422.csv' > ~/calibration_data_y_20231031_134422.csv
```

closed 2023-11-16 by Guilouz