Guilouz/Creality-Helper-Script-Wiki #92: Kamp issue after install

mindkill666, 2023-11-01:
When adding the kamp  update in moonraker.conf it displays a message when restarting

[update_manager KAMP for K1 Series]: Moonraker is not permitted to restart service 'KAMP for K1 Series'. To enable management of this service add KAMP for K1 Series to the bottom of the file /usr/data/printer_data/moonraker.asvc. To disable management for this service set 'is_system_service: False' in the configuration for this section.

Am i doing something wrong?

Guilouz, 2023-11-01:
Just add this:

# Remove '#' if you use KAMP
[update_manager KAMP for K1 Series]
type: git_repo
channel: dev
path: /usr/data/KAMP-for-K1-Series
origin: https://github.com/Guilouz/KAMP-for-K1-Series.git
primary_branch: main

mindkill666, 2023-11-01:
I just wanted to write this, i deleted the line # Remove '#' if you use KAMP
Works now :)

closed 2023-11-01 by Guilouz

pollinolas, 2023-11-02:
sorry i have the same issue,can´t fix it
Is it necessary to edit the /usr/data/printer_data/moonraker.asvc file?
my config looks like this:
[update_manager KAMP for K1 Series]
type: git_repo
channel: dev
path: /usr/data/KAMP-for-K1-Series
origin: https://github.com/Guilouz/KAMP-for-K1-Series.git
primary_branch: main

Guilouz, 2023-11-02:
> > Just add this:
> > # Remove '#' if you use KAMP
> > [update_manager KAMP for K1 Series] type: git_repo channel: dev path: /usr/data/KAMP-for-K1-Series origin: https://github.com/Guilouz/KAMP-for-K1-Series.git primary_branch: main
> 
> sorry i have the same issue,can´t fix it Is it necessary to edit the /usr/data/printer_data/moonraker.asvc file? my moonraker config looks like this:
> 
> [update_manager KAMP for K1 Series] type: git_repo channel: dev path: /usr/data/KAMP-for-K1-Series origin: https://github.com/Guilouz/KAMP-for-K1-Series.git primary_branch: main
> 
> i solved it adding this line: managed_services: klipper moonraker **UPDATE** but the KAMP folder comes out with a lock and the files cannot be edited

managed_services: klipper moonraker change nothing because klipper on K1 can't be manager, I have fixed this by:

  ```
  # Remove '#' from the lines below if you use KAMP
  [update_manager KAMP-for-K1-Series]
  type: git_repo
  channel: dev
  path: /usr/data/KAMP-for-K1-Series
  origin: https://github.com/Guilouz/KAMP-for-K1-Series.git
  primary_branch: main
  is_system_service: False
  ```
  
 The KAMP folder is read-only, this is normal, macros are not supposed to be modified (like original KAMP).