Guilouz/Creality-Helper-Script-Wiki #94: EOL error after  updating kamp[BUG] <title>

mindkill666, 2023-11-03:
### What happened?

After updating kamp i get an eol error on line 1.
I found it is in the smartpark.cfg, i had to comment the smart park function in kamp.cfg to be able to run.

### What did you expect to happen instead?

expected to work normal

### How to reproduce this issue?

update kamp in the update menu in fluidd, got eol error on firmware restart

mindkill666, 2023-11-03:
[image]

MuaiyadTSB, 2023-11-03:
ssh into your printer and modify line 13,14 in /usr/data/KAMP-for-K1-Series/Configuration/Smart_Park.cfg 

from: 

{% set center_x = bed_x_max / 2 | %}
{% set center_y = bed_y_max / 2 | %}

to:

{% set center_x = bed_x_max / 2 | float %}
{% set center_y = bed_y_max / 2 | float %}

Guilouz, 2023-11-03:
A mistake. It's fixed. Thanks.

closed 2023-11-03 by Guilouz