Guilouz/Creality-Helper-Script-Wiki #95: [BUG] <title>K1 KAMP Update Moonraker warning

Myklt8, 2023-11-06:
### What happened?

Wh`en I try to restart Moonraker
Moonraker warnings found.
[update_manager]: Failed to load extension KAMP-for-K1-Series: Invalid path for option `path` in section [update_manager KAMP-for-K1-Series]: Path `/usr/data/KAMP-for-K1-Series` does not exist
Unparsed config option 'origin: https://github.com/Guilouz/KAMP-for-K1-Series.git' detected in section [update_manager KAMP-for-K1-Series]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
Unparsed config option 'primary_branch: main' detected in section [update_manager KAMP-for-K1-Series]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.

### What did you expect to happen instead?

The update paths would install

### How to reproduce this issue?

Add any of these to the moonraker file and I get errors
[update_manager]
enable_auto_refresh: True
refresh_interval: 24
enable_system_updates: False

# Remove '#' from the lines below if you use Fluidd
[update_manager fluidd]
type: web
channel: beta
repo: fluidd-core/fluidd
path: /usr/data/fluidd

# Remove '#' from the lines below if you use KAMP
[update_manager KAMP-for-K1-Series]
type: git_repo
channel: dev
path: /usr/data/KAMP-for-K1-Series
origin: https://github.com/Guilouz/KAMP-for-K1-Series.git
primary_branch: main
is_system_service: False

Guilouz, 2023-11-06:
Answer is in error: Path/usr/data/KAMP-for-K1-Series` does not exist

That's mean you try to add KAMP to update manager but you have not KAMP installed in /usr/data/

Myklt8, 2023-11-06:
so the install helper script does not install in usr/data? How do I do that.. I mean what do I need to put in that section?

Myklt8, 2023-11-06:
[image]

Guilouz, 2023-11-06:
You don’t use latest script version!
Uninstall kamp and update script and reinstall kamp

Myklt8, 2023-11-06:
It fixed it and KAMP works..,.thank you so much!!!!

closed 2023-11-06 by Guilouz