Guilouz/Creality-Helper-Script-Wiki #100: KAMP line purge ignoring supports.

grrminator, 2023-11-08:
### What happened?

[image]

Image here. I know I can change the distance from the model in the CFG. But why does it ignore the support placement (the rings) there?

Is my only option editing the CFG?

### What did you expect to happen instead?

KAMP to know there is a support there

### How to reproduce this issue?

Use a model with support placement where the line purge MAY occur

grrminator, 2023-11-08:
Like id prefer not to edit the CFG so I can keep it at 10 if I printa big model. Shouldnt the purge be moveable or a little smarter? or does it genuinely not "know" a support is there?

Guilouz, 2023-11-08:
I think KAMP is not the problem here because KAMP uses the exclude_object function which indicates the dimensions of the model, this is given by the slicer.

What slicer do you use?

grrminator, 2023-11-08:
Orcaslicer

grrminator, 2023-11-08:
That circle is NOT the model. The purge line IS 10mm from the model. That circle is the first layer of a tree there

Guilouz, 2023-11-08:
Yes I have understand but slicer include supports to define area for cancel object.

posibble annissue in orcaslicer build, wich version do you use?

maybe you can try with another slicer to see.

grrminator, 2023-11-08:
I could check but would have to generate a profile

grrminator, 2023-11-08:
I could also give you the model to slice and see if it does it with you?

grrminator, 2023-11-08:
https://www.thingiverse.com/thing:6285690/files

grrminator, 2023-11-08:
https://github.com/SoftFever/OrcaSlicer/issues/2671 I raised an issue in orcaslicer based off this guys discussion. It appears it MAY be orca slicer. Which is disappointing.

Noisyfox, 2023-11-09:
If KAMP reads the boundary directly from EXCLUDE_OBJECT_DEFINE then that's the problem as the outline given from this command contains only the model outline but not support. Not sure if this is standard since PrusaSlicer does not include support in this command as well.

grrminator, 2023-11-09:
"The wipe/prime tower shouldn't be exposed to Klipper object exclusion feature. KAMP get printable size via object cancellation definition is hacky and against the intended use of object cancellation IMHO. It would be better to pass the minimum and maximum positions from slicer to it if KAMP supported, as is done with other adaptive bed mesh methods.  You might want to consider using https://github.com/eamars/klipper_adaptive_bed_mesh or https://github.com/Turge08/print_area_bed_mesh for better adaptive bed mesh management"

Per @Noisyfox above, this is the only way to fix it but im not sure how to do so with the installer script and this particular printer.

closed 2023-12-02 by Guilouz

grrminator, 2023-12-02:
Sorry to bug. But was this addressed?

DaMan9K, 2024-01-24:
Hi,In general, you can change the value of the parameter in KAMP_setting.cfg
parameter: variable_purge_margin is in fact a purge line and what difference does it make how far it is from the calibration area