Guilouz/Creality-Helper-Script-Wiki #101: Backup not working

lvlacleod, 2023-11-08:
### What happened?

Backup creates a file backup_config.tar but within it is 1 file called backup_config which is 77.5KB

### What did you expect to happen instead?

I expected it to back up all klipper files and KAMP files and add then to the tar file

### How to reproduce this issue?

Just run backup from the installer menu

Guilouz, 2023-11-08:
Backup is working fine. Make sure your root folder is not full.

lvlacleod, 2023-11-08:
Root folder is empty
[image]

This is what is in the tar file

Guilouz, 2023-11-09:
That's mean your root folder is full. You must restore factory settings.

lvlacleod, 2023-11-09:
Unless I am missing something obvious this is not full
[image]

Guilouz, 2023-11-10:
in SSH enter this command : df -H

And show me the result.

lvlacleod, 2023-11-10:
wow, no need to show results. size 107M used 107M but here it is anyway.

closed 2023-11-10 by lvlacleod