Guilouz/Creality-Helper-Script-Wiki #97: Movement out of Range

DasDing99, 2023-11-08:
### What happened?

Dear community,
since i made all the improvements provided i get this message after every attempt of printing
[image]

this issue only occurse when placing objects far in the front funnily enough

### What did you expect to happen instead?

The expectation was that it would be possible to print at every spot on the bed

### How to reproduce this issue?

I made all the steps given in the helper script, but it can be that I have overseen sth

Guilouz, 2023-11-08:
Need more info, stl, file used, gcode etc.

DasDing99, 2023-11-09:
Problem solved, reseted the printer and reinstalled everything and now all is fine

closed 2023-11-09 by Guilouz