Guilouz/Creality-Helper-Script-Wiki #103: [BUG] Error using KAMP with bicubic algorithm

jeanfbrito, 2023-11-10:
### What happened?

When we add more `probe_count` than 6,6, KAMP change to bicubic instead of lagrange. But then when it finishes the probing it makes klipper stop with this error:
```
23:49:24 // probe at 44.930,123.350 is z=0.375125
23:49:26 // Klipper state: Shutdown
23:49:26 !! {"code":"key60", "msg":"Internal error on command:_BED_MESH_CALIBRATE", "values": ["_BED_MESH_CALIBRATE"]}
23:49:27 !! {"code":"key60", "msg":"Internal error on command:BED_MESH_CALIBRATE", "values": ["BED_MESH_CALIBRATE"]}
23:49:27 !! {"code":"key60", "msg":"Internal error on command:START_PRINT", "values": ["START_PRINT"]}
23:50:43 $ FIRMWARE_RESTART
23:50:43 // Klipper state: Disconnect
```

As I was trying to use bicubic instead of lagrange as mentioned here: https://github.com/Guilouz/Creality-K1-and-K1-Max/wiki/Improve-Bed-Mesh I started to get this errors, so I need to keep low probe_count to make KAMP not change to bicubic algo.

I don't know if this is a bug on K1 Max Klipper or what. I'm trying to improve the bed leveling because there are some points in the bed that the filament don't stick well at the second layer.

### What did you expect to happen instead?

Work without errors.

### How to reproduce this issue?

Increase `probe_count` numbers until KAMP start using `bicubic` algorithm.

My config:
```
[bed_mesh]
speed: 150
mesh_min: 10,10
mesh_max: 290,290
probe_count: 15,15
fade_start: 3.0
fade_end: 10.0
algorithm: bicubic
bicubic_tension: 0.1
```

### Additional information:

I'm using a K1 Max with the firmware `1.3.2.1`. 
The Helper Script and KAMP are updated from today.

lvlacleod, 2023-11-10:
I too wanted to use bicubic so all i did was change this

    {% if (([points_x, points_y]|max) > 6) %}    
        {% set algorithm = "bicubic" %}  
        {% set min_points = 4 %}
    {% else %}  
        {% set algorithm = "lagrange" %} 
        {% set min_points = 3 %}  
    {% endif %}

To this

    {% if (([points_x, points_y]|max) > 6) %}    
        {% set algorithm = "lagrange" %}  
        {% set min_points = 4 %}
    {% else %}  
        {% set algorithm = "bicubic" %} 
        {% set min_points = 3 %}  
    {% endif %}

Guilouz, 2023-11-10:
Creality OS doesn't support more than 6x6 bed mesh.

closed 2023-11-10 by Guilouz

jeanfbrito, 2023-11-10:
> Creality OS doesn't support more than 6x6 bed mesh.

Any idea if we can fix this? Is their klipper that blocks this?